sandeepmistry/arduino-LoRa #108: Addresses Limit with LoRa Duplex

BlackxRyan, 2018-02-21:
I'm using a little modified code of LoRa Duplex with 2 Arduino cards + 2 Dragino LoRa Shields. The code works correctly.

I just have a question about the addresses. With this model, you can have 35 different addresses (without counting the broadcast) and I want to know if you can overpass this limit.

`#include <SPI.h>              // include libraries
#include <LoRa.h>
//
//const int csPin = 7;          // LoRa radio chip select
//const int resetPin = 6;       // LoRa radio reset
//const int irqPin = 1;         // change for your board; must be a hardware interrupt pin

String outgoing;              // outgoing message

byte msgCount = 0;            // count of outgoing messages
byte localAddress = 0xab;     // address of this device
byte destination = 0xbc;      // destination to send to
long lastSendTime = 0;        // last send time
int interval = 5000;          // interval between sends

void setup() {
  Serial.begin(9600);                   // initialize serial
  while (!Serial);

  Serial.println("LoRa Duplex");

//  // override the default CS, reset, and IRQ pins (optional)
//  LoRa.setPins(csPin, resetPin, irqPin);// set CS, reset, IRQ pin

  LoRa.begin(868E6);

  Serial.println("LoRa init succeeded.");
}

void sendMessage(String outgoing) {
  LoRa.beginPacket();                   // start packet
  LoRa.write(destination);              // add destination address
  LoRa.write(localAddress);             // add sender address
  LoRa.write(msgCount);                 // add message ID
  LoRa.write(outgoing.length());        // add payload length
  LoRa.print(outgoing);                 // add payload
  LoRa.endPacket();                     // finish packet and send it
  msgCount++;                           // increment message ID
}

void onReceive(int packetSize) {
  if (packetSize == 0) return;          // if there's no packet, return

  // read packet header bytes:
  int recipient = LoRa.read();          // recipient address
  byte sender = LoRa.read();            // sender address
  byte incomingMsgId = LoRa.read();     // incoming msg ID
  byte incomingLength = LoRa.read();    // incoming msg length

  String incoming = "";

  while (LoRa.available()) {
    incoming += (char)LoRa.read();
  }

  if (incomingLength != incoming.length()) {   // check length for error
    Serial.println("error: message length does not match length");
    return;                             // skip rest of function
  }

  // if the recipient isn't this device or broadcast,
  if (recipient != localAddress && recipient != 0xFF) {
    Serial.println("This message is not for me.");
    return;                             // skip rest of function
  }

  // if message is for this device, or broadcast, print details:
  Serial.println("Received from: 0x" + String(sender, HEX));
  Serial.println("Sent to: 0x" + String(recipient, HEX));
  Serial.println("Message ID: " + String(incomingMsgId));
  Serial.println("Message length: " + String(incomingLength));
  Serial.println("Message: " + incoming);
  Serial.println("RSSI: " + String(LoRa.packetRssi()));
  //Serial.println("Snr: " + String(LoRa.packetSnr()));
  Serial.println();
}

void loop() {
  if (millis() - lastSendTime > interval) {
    String message = "Bonjour";   // send a message
    sendMessage(message);
    Serial.println("Sending " + message);
    lastSendTime = millis();            // timestamp the message
    interval = 5000;    // 2-3 seconds
  }

  // parse for a packet, and call onReceive with the result:
  
  onReceive(LoRa.parsePacket());
}`

morganrallen, 2018-02-21:
Why are you only using only using `A-F`?
The Duplex example (https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaDuplex/LoRaDuplex.ino) allows 254 addresses (`0x00 - 0xFE`) with `0xFF` being the broadcast. The important thing to note here is the radios themselves have no concept of addressing. This example is simply utilizing a fixed byte position in the incoming packet as an address. With this, you could even use more than 1 byte if you wanted.

BlackxRyan, 2018-02-22:
Oh ok, I didn't understand it correctly.

Thanks for the help.

halukmy, 2018-02-22:
@morganrallen  what if up then 255 address?

morganrallen, 2018-02-22:
All of the addressing logic is found right in that example code. The library provides no additional code to handle addresses.

```
if (recipient != localAddress && recipient != 0xFF) {
...
}
```

That `if` checks if the address is yours or broadcast (`0xFF`). So if you wanted 255 addresses you could drop broadcast or move to a 16 bit address space.

BlackxRyan, 2018-02-23:
Hum, when I use addresses with numbers like 0x00 or 0x55, both of them use the same address and I don't know why.
[image]

sandeepmistry, 2018-02-26:
Closing this as the original question was solved by @morganrallen in https://github.com/sandeepmistry/arduino-LoRa/issues/108#issuecomment-367734415.

@BlackxRyan please open a new issue for the problem you mention above. Please include full sketches and/or details to reproduce.

closed 2018-02-26 by sandeepmistry