sandeepmistry/arduino-LoRa #110: question : arduino-LoRa vs LoraWan

halukmy, 2018-02-23:
everybody please to add their thinks 

cons and pros this repo vs lorawan ?

first let me add

This Repo Pro : no need gateway
               cons : every node listen every other node

BlackxRyan, 2018-02-23:
By the way can you use this library with LoRa Heltec and Seeeduino LoRaWAN ?

wero1414, 2018-02-25:
With LoRAWAN you have Address of each device and the most important thing, SECURITY, cause when you transmit a packet with LoRA as any modulation else, any radio at the same frequency will receive the packet, so its important to encrypt in a safe way your data, another important thing in LoRaWAN is that you have a complete network where you can manage a lot of devices with different frequencies based on the sub-channel you select, with the frequencies multiplexed like that you'll have a more free spectrum to make communication more affordable.

LoRA by itself work as a M2M communication where you can send information with very low power by a great distance or as well in a noisy enviroment (speaking electromagnetically), it allows you that by the modulation nature

sandeepmistry, 2018-02-26:
@wero1414 thanks for answering. Doesn't LoRaWAN encrypt data packets as back for the protocol stack?

Is this something to PR to the read me or a new FAQ document?

closed 2018-04-08 by sandeepmistry