sandeepmistry/arduino-LoRa #114: LoRa Duplex : problem with 0x*number* communication

BlackxRyan, 2018-02-26:
Hello.

I'm using a little modified code of LoRa Duplex with 2 Arduino cards + 2 Dragino LoRa Shields. The code works correctly.

But when I use addresses with numbers like 0x15 or 0x55, both of them use the same address and I don't know why. However, I'm compiling it on two different Arduino cards.

[image]

Here's the modified code I'm using :

```
#include <SPI.h>              // include libraries
#include <LoRa.h>
//
//const int csPin = 7;          // LoRa radio chip select
//const int resetPin = 6;       // LoRa radio reset
//const int irqPin = 1;         // change for your board; must be a hardware interrupt pin

String outgoing;              // outgoing message

byte msgCount = 0;            // count of outgoing messages
byte localAddress = 0x22;     // address of this device
byte destination = 0x23;      // destination to send to
long lastSendTime = 0;        // last send time
int interval = 5000;          // interval between sends

void setup() {
  Serial.begin(9600);                   // initialize serial
  while (!Serial);

  Serial.println("LoRa Duplex");

//  // override the default CS, reset, and IRQ pins (optional)
//  LoRa.setPins(csPin, resetPin, irqPin);// set CS, reset, IRQ pin

  LoRa.begin(868E6);

  Serial.println("LoRa init succeeded.");
}

void sendMessage(String outgoing) {
  LoRa.beginPacket();                   // start packet
  LoRa.write(destination);              // add destination address
  LoRa.write(localAddress);             // add sender address
  LoRa.write(msgCount);                 // add message ID
  LoRa.write(outgoing.length());        // add payload length
  LoRa.print(outgoing);                 // add payload
  LoRa.endPacket();                     // finish packet and send it
  msgCount++;                           // increment message ID
}

void onReceive(int packetSize) {
  if (packetSize == 0) return;          // if there's no packet, return

  // read packet header bytes:
  int recipient = LoRa.read();          // recipient address
  byte sender = LoRa.read();            // sender address
  byte incomingMsgId = LoRa.read();     // incoming msg ID
  byte incomingLength = LoRa.read();    // incoming msg length

  String incoming = "";

  while (LoRa.available()) {
    incoming += (char)LoRa.read();
  }

  if (incomingLength != incoming.length()) {   // check length for error
    Serial.println("error: message length does not match length");
    return;                             // skip rest of function
  }

  // if the recipient isn't this device or broadcast,
  if (recipient != localAddress && recipient != 0xFF) {
    Serial.println("This message is not for me.");
    return;                             // skip rest of function
  }

  // if message is for this device, or broadcast, print details:
  Serial.println("Received from: 0x" + String(sender, HEX));
  Serial.println("Sent to: 0x" + String(recipient, HEX));
  Serial.println("Message ID: " + String(incomingMsgId));
  Serial.println("Message length: " + String(incomingLength));
  Serial.println("Message: " + incoming);
  Serial.println("RSSI: " + String(LoRa.packetRssi()));
  //Serial.println("Snr: " + String(LoRa.packetSnr()));
  Serial.println();
}

void loop() {
  if (millis() - lastSendTime > interval) {
    String message = "Bonjour";   // send a message
    sendMessage(message);
    Serial.println("Sending " + message);
    lastSendTime = millis();            // timestamp the message
    interval = 5000;    
  }

  // parse for a packet, and call onReceive with the result:
  
  onReceive(LoRa.parsePacket());
}
```

WillingLearner, 2018-06-29:
Hello ryan I am having the same difficulty with address and cant understand completely the main idea of duplex communication. I cant make the codes work yet. I need the help. When I read this issue I think I understand my problem.

I upload the same sketch into the 2 arduino and then doesn't work. Does it need to be like the following ?

For first sketch : 
...
byte localAddress = 0xab;     // address of this device
byte destination = 0xbc;      // destination to send to
...

For second sketch : 
..
byte localAddress = 0xbc;     // address of this device
byte destination = 0xab;      // destination to send to
..

WillingLearner, 2018-06-29:
Here is my codes for first and second device :

First :

`#include <SPI.h>
#include <LoRa.h>

int msgCount = 0;
int local = 2; //ID of this device
int dest = 1; //destination to send to
long lastSendTime = 0;
int interval = 2000; 
String incoming; 
String outgoing; 

void setup() 
{
Serial.begin(115200);
while (!Serial);
LoRa.begin(433E6);
}

void loop() 
{
if (millis() - lastSendTime > interval)
{
String message = "Hello";
sendMessage(message);
Serial.println("Sending " + message);
Serial.println("");
lastSendTime = millis();
interval = 2000; 
}

int packetSize = LoRa.parsePacket();

if (packetSize) 
{
int localAddress = LoRa.read();
int dest = LoRa.read();
int id = LoRa.read();

if (localAddress == 1) 
{
Serial.print("LoRa ");
Serial.println(localAddress);
Serial.print("For LoRa ");
Serial.println(dest);
Serial.print("ID = ");
Serial.println(id);
Serial.print("Message : ");
while (LoRa.available()) {
incoming = (char)LoRa.read();
Serial.print(incoming);
}

Serial.println("");
Serial.print("RSSI : ");
Serial.println(LoRa.packetRssi());
Serial.println("");
}
}
}

void sendMessage(String outgoing)
{
LoRa.beginPacket();
LoRa.write(local);
LoRa.write(dest);
LoRa.write(msgCount);
LoRa.print(outgoing);
LoRa.endPacket();
msgCount++;
}`

Second : 

`#include <SPI.h>
#include <LoRa.h>

int msgCount = 0;
int local = 1; //ID of this device
int dest = 2; //destination to send to
long lastSendTime = 0;
int interval = 2000; 
String incoming; 
String outgoing; 

void setup() 
{
Serial.begin(115200);
while (!Serial);
LoRa.begin(433E6);
}

void loop() 
{
if (millis() - lastSendTime > interval)
{
String message = "Hello";
sendMessage(message);
Serial.println("Sending " + message);
Serial.println("");
lastSendTime = millis();
interval = 2000; 
}

int packetSize = LoRa.parsePacket();

if (packetSize) 
{
int localAddress = LoRa.read();
int dest = LoRa.read();
int id = LoRa.read();

if (localAddress == 2) 
{
Serial.print("LoRa ");
Serial.println(localAddress);
Serial.print("For LoRa ");
Serial.println(dest);
Serial.print("ID = ");
Serial.println(id);
Serial.print("Message : ");
while (LoRa.available()) {
incoming = (char)LoRa.read();
Serial.print(incoming);
}

Serial.println("");
Serial.print("RSSI : ");
Serial.println(LoRa.packetRssi());
Serial.println("");
}
}
}

void sendMessage(String outgoing)
{
LoRa.beginPacket();
LoRa.write(local);
LoRa.write(dest);
LoRa.write(msgCount);
LoRa.print(outgoing);
LoRa.endPacket();
msgCount++;
}`

Uploaded the code into arduinos. System works randomly. Sometimes sending data for once sometimes for 5 times . By luck Data is came for once :) I dont understand anything why

WillingLearner, 2018-06-30:
I tried what you suggested literally but still it doesn't work. This is what happened to my system :  

I reset first arduino and approximately 0.5 sn later second arduino that connected to serial monitor. For this scenario, I can receive message truly but for once . If I firstly reset second arduino that connected to serial monitor, message cannot be taken but just sending. What I understand from this behavior of system is that There is some kind of trouble in shifting into either sender mode or receiver mode. It seems One arduino cannot have both ability somehow. 

What are the possible mistake I can make ?

My wire connections seem true because I can perform LoraReceiver and LoraSender example successfully it works perfectly I got very good result like communicating from 7km far away for now.

BlackxRyan, 2018-06-30:
No problem, glad I could help you :)

closed 2018-06-30 by BlackxRyan

halukmy, 2018-07-01:
@BlackxRyan what if device node total is over 255?

i guess request page is only byte value 0 to 255

can i send 1055 and more devices?

what shoul di change?

BlackxRyan, 2018-07-02:
I don't understand your question.

thannara123, 2020-08-11:
How clear it , iam facing same issue ?

thannara123, 2020-08-11:
I figure out the trouble. I change Arduinos and system start to work properly fortunately.  i didnt understand

hmhuang0501, 2020-12-21:
@thannara123 Can you elaborate more how you solve the issue? I did sequentially/one-by-one compile the 2 sketch for 2 LoRa boards separately, but both of them still do the sending task, no receiving at all....