sandeepmistry/arduino-LoRa #115: Starting LoRa failed error

kayani009, 2018-02-26:
i am usinf SX1276 Lora module. but I gives Starting LoRa failed error. can you help please?

halukmy, 2018-02-26:
check damn pins

morganrallen, 2018-02-26:
@halukmy that is not helpful.

@kayani009 First thing to check is that the pins are set correctly. Be default the pins are.
```
Slave Select: 10
Reset:             9
DIO0:             2
```
If your dev board uses anything else, you can change with `setPins`

More reading here: https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-pins

kayani009, 2018-02-26:
I am not using any gate way. There is no involement of Internet gateway.
Its simple transmitter and receiver.

Joypacharya, 2018-02-28:
I am using RA- 01 module. I connected this Lora module with my Uno. But all time it shows Lora failed !!
So, help me out this problem.

aniketmad, 2018-03-31:
we are also facing problem of Starting lora failed....!

please please please  help me out!!!

morganrallen, 2018-04-06:
@kayani009 @Joypacharya @aniketmad 
If you have not read about `setPins` please read this.
https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-pins
Then check and double check that all you're pins are set correctly.
If you are still having problems, please describe what board you are using, including; which pins are connected to which signals.

mark9599, 2019-06-04:
I've rechecked my pins, changed Arduino Boards, still doesn't work.
Can you suggest any fix?