sandeepmistry/arduino-LoRa #125: No data received between ra-01's

dwaynez, 2018-03-21:
I have 2 ra01 Loras connected to 2 nano clones. I am using a logic leveler and and external power supply to the Loras. I run the sample duplicate sketch and all I ever get I the messages about the message being sent but nothing is ever received. My questions:

Would lowering the spi frequency help even though I am able to dump registers ok? (I am away from my bench and unable to fry for a few weeks).

Everything seems to be connected ok and I am able to initialize and dump registers. Could the problem be on the receive side or the transmit side and any suggestions for debugging?

The logic converter I have is only 4 channels so I don't have the reset pin connected. Is this a problem?

Any ideas on what I could try to resolve my non communication?

dwaynez, 2018-03-21:
That should be duplex sketch.

sandeepmistry, 2018-04-08:
I believe the RA01's have the RFO pin connected instead of PA boost.

Try the info from: https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#tx-power
```
LoRa.setTxPower(14, PA_OUTPUT_RFO_PIN)
```

bharath-krishnaswamy, 2018-04-10:
What about the LoRa Ra-02 modules. The same problem is occurring to me as well!!

thomashamain, 2020-03-27:
@dwaynez, did you manage to solve your problem? I'm running into a similar issue.
Both of my Lora modules seem to work, they respond to register dumps. The data I want to transmit ("hello" plus de counter of the Arduino sketch example) are correctly sent to the Sender side. However, at the receiving end, I observe strange things. Sometimes get simply "Received packet '' wit RSSI -164", so without any payload. The positive side is that it is in synch with the sender. Other times I get strange responses as seen in the screenshot. The most interesting is that when I play around with ground cables, I switch between the two kinds of responses.
[image]

dwaynez, 2020-03-27:
I was able to get everything working. The biggest problem I had was that after a write operation, the contents of REG_MODEM_CONFIG_1 were getting getting changed, so when I looped around to send again, things hung up because the LORA was not in the correct state. I had to check REG_MODEM_CONFIG_1 before every write and reset it if necessary.

I vaguely recall seeing what you are seeing on your output, but I am afraid I cannot remember what the issue was or what I did to resolve it. I can say that the LORA's are rock solid in that they do not garble transmission. I found it very effective to go into the LoRa.cpp file and put in debug statements.

thomashamain, 2020-03-28:
Thanks for your advice @dwaynez .
Indeed, the value in REG_MODEM_CONFIG_1 is fluctuating for the receiver and the sender. I inserted the following lines just before beginPacket(). I rendered writeRegister as a public function for convenience reasons, and modified dumpRegisters to display only the register 0x1D. The result is that I don't get any printf from dumpRegisters, meaning I managed to force 0x72 for the sender.
```cpp
while(LoRa.readRegister(0x1D)!=0x72)
{
    LoRa.writeRegister(0x1D,0x72);
    LoRa.dumpRegisters(Serial);
  }
```
For the receiver, I did the same, but it keeps changing all the time... I suspect an issue with the power supply as you mentioned in the issue [.](https://github.com/sandeepmistry/arduino-LoRa/issues/191#issuecomment-428774076). I have a 5A capable variable power supply set on 3.3V. I have put a 47uF capacitor to try and level out the voltage fluctuation but it did not help.
I feel that I am close because I once or twice saw a "Received 'rello' with RSSI -100", so it is moving in the write direction. 😃

IoTThinks, 2020-03-28:
REG_MODEM_CONFIG_1 seems to be signal bandwidth, right?
REG_MODEM_CONFIG_1 = 0x72

0x72 >> 4 is 7 => 125E3 (125Khz)
So can we set setSignalBandwidth(125E3) before writing so it will basically do the same thing?
Of course, writeRegister is faster and lighter.

```
long LoRaClass::getSignalBandwidth()
{
  byte bw = (readRegister(REG_MODEM_CONFIG_1) >> 4);

  switch (bw) {
    case 0: return 7.8E3;
    case 1: return 10.4E3;
    case 2: return 15.6E3;
    case 3: return 20.8E3;
    case 4: return 31.25E3;
    case 5: return 41.7E3;
    case 6: return 62.5E3;
    case 7: return 125E3;
    case 8: return 250E3;
    case 9: return 500E3;
  }

  return -1;
}
```

thomashamain, 2020-03-28:
Yes, this is correct, and it is as well the Coding Rate and if the message has an implicit or explicit header (implicit = the receiver know exactly the byte length of the message so no need for a header, explicit = the message has a header with the size of the message, CRC and other info)
[image]
The issue is that if the registers on the sending and receiving ends does not match, I don't see how they will be able to communicate. And, in my case, the register REG_MODEM_CONFIG_1 changes randomly on the receiving side...

thomashamain, 2020-03-28:
So I resolved my issue by disconnecting the grounds of my Arduinos from the LoRa modules. Since my power supply and my PC is grounded to the same network, this works. However, I don't know if it will on a battery with the same ground. And I don't know why there is a conflict in the first place...
I can say that forcing the REG_MODEM_CONFIG_1 to a desired value helped as well because I already tried to communicate with the grounds connected but without writing in the registers and it didn't work. Thanks for your time.

olehpratsko, 2022-09-17:
Maybe my experience will help somebody. I haven't checked my registers states. But on the receiver side I had Ra-02 connected to 29 pin Arduino Nano ground. Receiver looked to stop receiving packages after 20-40 seconds. After reading the last [thomashamain](https://github.com/thomashamain) comment, I've connected Ra-02 ground to pin 4 ground, and it looks like connection doesn't fail anymore.