sandeepmistry/arduino-LoRa #126: Question: my LoRa can work in freq 915MHz but its a 433Mhz module. why?

Black3rror, 2018-03-27:
Hi all.
I'm new to LoRa. sorry if my question isn't relative to repository but I didn't know where to ask.
I have a LoRa-Ra02 (SX1278) that works with arduino UNO. something like this : 
[image]
my question is that as u can see on its cover (in the image above) its for 410 to 525 Mhz but I've tested it in freq 915Mhz and 390Mhz and it worked.
can I use it in any freq? or I should buy a LoRa for those freqs? at all what does ISM band on the cover say?

halukmy, 2018-03-27:
no

just 433

Black3rror, 2018-03-29:
thanx @halukmy 
but why I can work on 390MHz. I thought maybe the chip changes it to minimum freq (410MHz) but I've checked the register and it was on 390MHz and even I've tested on 390MHz and 391MHz and it was difference between them so the chip doesn't change the freq.
so whats happening?

halukmy, 2018-03-30:
it works but very very low communication quailty

you need to touch antennas,, i was like that

closed 2018-04-08 by sandeepmistry