sandeepmistry/arduino-LoRa #127: Lora onReceive Callback not triggered,  but data received in loop()

acbrandao, 2018-03-27:
Strange behavior, I tried altering the LoRaDuplexCallback example, integrating it into my TTGO Lora board. Sender is sending as expected, and data is reaching my receiver node, but the Callback is not getting triggered.  Code segment looks like this..

`setup()
{
...
  LoRa.onReceive(onReceive);  //register the Callback
  LoRa.receive();
  Serial.println("LoRa init succeeded.");
...
}

void onReceive(int packetSize) {
...
 //IT NEVER prints this line or any other code    
  Serial.println("Received packet size:"+(String)packetSize );
... 
} `

If however I place the receive inside the main Loop() as below it works?
`
 if (packetSize)     // received a packet
    {
     int  rssi = LoRa.packetRssi();
    
    Serial.print("Received packet '");
    Serial.println("Packet  RCVD "+String(packetSize)+"bytes" );
    
    // read packet
    String packet="";                   // Clear packet
    while (LoRa.available()) 
    packet += (char)LoRa.read(); // Assemble new packet    
    Serial.println(packet);
  }`

Do I need to do anything else beside register the OnReceive for it to work?

Hardware: TTGO Lora OLED v.1

acbrandao, 2018-03-27:
Ok I tried other things like..

``` 
 LoRa.setPins(LORA_NSS, LORA_NRESET, LORA_DEFAULT_DIO0_PIN);

```
This did not help the issue, no change.

``` //pinMode(LORA_DEFAULT_DIO0_PIN, 99); ```
Caused a stack overflow  and reboot..

torntrousers, 2018-03-27:
According to [this](https://primalcortex.wordpress.com/2017/11/24/the-esp32-oled-lora-ttgo-lora32-board-and-connecting-it-to-ttn/) for the TTGO board LORA NSS is pin 18, reset is 14 and i think DIO0 is what that refers to as LoRa-IRQ so is pin 26. So try ```LoRa.setPins(18, 14, 26)```?

MarcRWetzel, 2018-04-02:
Hi,

I fixed this by setting the IRQ pin to INPUT. First I tested it in my own sketch, but I think this should be patched in the library, like so:

```
int LoRaClass::begin(long frequency)
{
  // setup pins
  pinMode(_ss, OUTPUT);
  pinMode(_reset, OUTPUT);
  pinMode(_dio0, INPUT); // added

[...]

```
BR
Marc

MarcRWetzel, 2018-04-07:
@morganrallen Aahrg, you are right. I had a short test, and it didn't worked. So I started fumbling... thought that I found the cause... I had opened another issue, where receiving stopped after some hours, and didn't know why. Changing the code over to use IRQs helped me a lot. But I had the IRQ pin always at pin 26...

closed 2018-04-08 by sandeepmistry

github-marcelo, 2021-12-31:
The LoRa.h Library woks fine!!!  This problem occurs when some parameters are wrongly defined. The correct for Module HELTEC ESP32 LORA ( version 01 and 02 ) is:

#define SCK_LORA              5
#define MISO_LORA             19
#define MOSI_LORA             27
#define RESET_PIN_LORA  14
#define CS_PIN_LORA         18
#define LORA_DEFAULT_DIO0_PIN 26
#define HIGH_GAIN_LORA     20  /* dBm */
#define BAND               915E6  /* 915MHz de frequencia */
// ...

/* Func: init comunic chip LoRa */
bool init_comunicacao_lora(void)
{
    bool status_init = false;
    Serial.println("[LoRa Receiver] Tentando iniciar comunicacao com o radio LoRa...");
    SPI.begin(SCK_LORA, MISO_LORA, MOSI_LORA, CS_PIN_LORA);
    LoRa.setPins(CS_PIN_LORA, RESET_PIN_LORA, LORA_DEFAULT_DIO0_PIN);
     
    if (!LoRa.begin(BAND)) 
    {
        Serial.println("[LoRa Receiver] Comunicacao com o radio LoRa falhou. Nova tentativa em 1 segundo...");        
        delay(1000);
        status_init = false;
    }
    else
    {
        /* Configura o ganho do receptor LoRa para 20dBm, o maior ganho possível (visando maior alcance possível) */ 
        LoRa.setTxPower(HIGH_GAIN_LORA); 
        Serial.println("[LoRa Receiver] Comunicacao com o radio LoRa ok");
        status_init = true;
    }
 
    return status_init;
}

DaveK20, 2023-08-12:
> The LoRa.h Library woks fine!!! This problem occurs when some parameters are wrongly defined. The correct for Module HELTEC ESP32 LORA ( version 01 and 02 ) is:
>
> #define SCK_LORA 5 #define MISO_LORA [...]

Voce poderia postar um codigo base com esas modificacoes funcionando?
Acabei de tentar com oque foi postado por voce e nao funcionou.

Alain-Godo, 2023-08-14:
Hi everyone!!! I have a related issue. When I receive the data in a sync way all is fine (using the loop for example) but the callback function is never triggered using the onReceive() method.

Board: *ESP32 DOIT DEVKIT V1*
Pin used: *HSPI + 4 (Reset) + 2 (DIO0)*

### Setup code:

hspi = new SPIClass(HSPI);
hspi->begin();
lora2.setSPI(*hspi);

lora2.setPins(15,4,2);
Serial.println("Begining 2...");
Serial.println(lora2.begin(915E6));
lora2.onReceive(lora2Callback);
lora2.receive();

### Callback function:

void lora2Callback(int packet_size) {
   Serial.print("Received packet: '");

    // read packet
    while (lora2.available()) {
    Serial.print((char)lora2.read());
    }

    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(lora2.packetRssi());
}