sandeepmistry/arduino-LoRa #129: RX current

turbyho, 2018-03-28:
Hi, 
I set power from default 17dB to 2dB and current consumption is still around 60mA (only radio). It looks like a problem with TxPower settings.

T

morganrallen, 2018-04-06:
Can you describe your board setup and how you're taking power measurements? 
I'll make a comparison on one of my dev boards and look into registry settings.

turbyho, 2018-04-06:
I think there is a problem with the PA turned on. I use this changes.

...
void LoRaClass::setMaxTxCurrent(int maxCurrent) {

  //  45-120mA
  if (maxCurrent < 45) {
    maxCurrent = 45;
  }
  if (maxCurrent > 120) {
    maxCurrent = 120;
  }
  unsigned int ocp = (maxCurrent - 45) / 5;
  writeRegister(REG_OCP, 0x20 | ocp);
}

void LoRaClass::setTxPower2( int8_t power )
{
  uint8_t RegPaConfing = readRegister(REG_PA_CONFIG);
  uint8_t RegPaDac = readRegister(REG_PA_DAC);

  if ( ( RegPaConfing & 0x80 ) == 0x80 )
  {
    if ( ( RegPaDac & 0x87 ) == 0x87 )
    {
      if ( power < 5 )
      {
        power = 5;
      }
      if ( power > 20 )
      {
        power = 20;
      }
      RegPaConfing = ( RegPaConfing & 0x8F ) | 0x70;
      RegPaConfing = ( RegPaConfing & 0xF0 ) | ( uint8_t )( ( uint16_t )( power - 5 ) & 0x0F );
    }
    else
    {
      if ( power < 2 )
      {
        power = 2;
      }
      if ( power > 17 )
      {
        power = 17;
      }
      RegPaConfing = ( RegPaConfing & 0x8F ) | 0x70;
      RegPaConfing = ( RegPaConfing & 0xF0 ) | ( uint8_t )( ( uint16_t )( power - 2 ) & 0x0F );
    }
  }
  else
  {
    if ( power < -1 )
    {
      power = -1;
    }
    if ( power > 14 )
    {
      power = 14;
    }
    RegPaConfing = ( RegPaConfing & 0x8F ) | 0x70;
    RegPaConfing = ( RegPaConfing & 0xF0 ) | ( uint8_t )( ( uint16_t )( power + 1 ) & 0x0F );
  }
  writeRegister( REG_PA_CONFIG, RegPaConfing );
}

void LoRaClass::setPa20dBm( bool enale )
{
  uint8_t RegPaConfing = readRegister(REG_PA_CONFIG);
  uint8_t RegPaDac = readRegister(REG_PA_DAC);

  if ( ( RegPaConfing & 0x80 ) == 0x80 )
  {
    if ( enale == true )
    {
      RegPaDac = 0x87;
    }
  }
  else
  {
    RegPaDac = 0x84;
  }
  writeRegister(REG_PA_DAC, RegPaDac);
  writeRegister(REG_PA_CONFIG, RegPaConfing);
}
...

sandeepmistry, 2018-04-28:
Closing this for now, due to lack of activity.

closed 2018-04-28 by sandeepmistry