sandeepmistry/arduino-LoRa #130: Lock-Up of receiver after several hours

MarcRWetzel, 2018-04-01:
Hi,
first of all - thanks for this nice little software for getting me on to the LoRa ship real fast.

I use the library on two TTGO wifi lora 32 boards (433MHz). 
Development environment for both the sender and receiver is sloeber.

My problem is, that the sender runs for days, but the receiver locks up every now and then. Sometimes he receives 4k packets, sometimes 50k but he will lock up eventually.

Is there some known issue with the esp32 and your code? Or the underlying libraries like SPI?
I am somehow new to the sloeber/Arduino approach - normally I would fire up my Openocd and debug, but I didnt get this to work on the esp32. 

Has somebody some clue, what I can do, to hunt this little bug down?

Here comes my little code fragment, which does not work (after several hours):

`#include <LoRa.h>

String receivedText;
String receivedRssi;

// WIFI_LoRa_32 ports
// GPIO5  -- SX1278's SCK
// GPIO19 -- SX1278's MISO
// GPIO27 -- SX1278's MOSI
// GPIO18 -- SX1278's CS
// GPIO14 -- SX1278's RESET
// GPIO26 -- SX1278's IRQ(Interrupt Request)

#define SS      18
#define RST     14
#define DI0     26
#define BAND    433E6

char timestring[20];
// LoRa stats
int rssi;
float snr;

// t is time in seconds = millis()/1000;
void timeToString(char* string, size_t size)
{
	unsigned long nowMillis = millis();
	unsigned long seconds = nowMillis / 1000;
	int days = seconds / 86400;
	nowMillis %= 1000;
	seconds %= 86400;
	byte hours = seconds / 3600;
	seconds %= 3600;
	byte minutes = seconds / 60;
	seconds %= 60;
	snprintf(string, size, "%02d:%02d:%02d:%02d.%03d", days, hours, minutes, (int)seconds,(int)nowMillis);
}

void setup() {

	SPI.begin(5, 19, 27, 18);
	LoRa.setPins(SS, RST, DI0);

	Serial.begin(115200);

	Serial.println("LoRa Receiver");

	if (!LoRa.begin(BAND)) {
		Serial.println("Starting LoRa failed!");
		while (1);
	}
}

int readPacket(void* buffer, size_t len)
{
	char buf[255];
	int i,rcv;

	i=0;
	rssi=-200;
	snr=100.;

	while (LoRa.available()) {
		rcv =  (char)LoRa.read();

		rssi = LoRa.packetRssi();
		snr = LoRa.packetSnr();
		//Serial.printf("RSSI:%d SNR:%2.2f\n",rssi,snr);

		if (rcv>=0) {
			buf[i++]= (char)rcv;
		} else {
			Serial.printf("readPacket: Error(%d) after %d bytes\n",rcv,i);
			return -1;
		}
		if (i>255) {
			Serial.printf("readPacket: Error(MAX_LEN) after %d bytes\n",i);
			return -2;
		}
	}

	Serial.printf("readPacket: Packet received with %d bytes, RSSI:%d, SNR:%2.2f\n",i,rssi,snr);
	if (i<=len)
		memcpy(buffer,buf,i);
	else {
		Serial.printf("readPacket: Error: Packet received with %d>%d bytes\n",i,len);
		return -3;
	}
	return i;
}

#define MRW_MAGIC_FIRST 0b10101010
#define MRW_MAGIC_LAST  0b01010101

struct Packet {
	uint8_t magic_first;
	uint8_t size;
	uint32_t id;
	uint8_t magic_last;
} p;

int parsePacket(void) {
	int result;

	result = readPacket((void*)&p,sizeof(p));
	if (result) {
		// check
		if (p.size != sizeof(p)) Serial.println("Packet: Size() mismatch");
		if (p.magic_first != MRW_MAGIC_FIRST) Serial.println("Packet: MAGIC_FIRST error");
		if (p.magic_last != MRW_MAGIC_LAST) Serial.println("Packet: MAGIC_LAST error");
		Serial.println("Id: "+(String)p.id);
	}
	return result;
}

void loop() {
	// try to parse packet
	int packetSize = LoRa.parsePacket();

	if (packetSize) {

		Serial.printf("LoRa.Parsepacket: %d\n",packetSize);

		if (parsePacket()) {
			Serial.printf("Id: %d   ",p.id);
		} else {
			Serial.printf("-ERROR-");
		}

		timeToString(timestring, sizeof(timestring));
		Serial.println(timestring);

	}
}
`

Please ignore the packet-struct. In my previous attempts I used just a simple hello+counter (as in the examples) which would break also.

Maybe somebody has some snippet, to get a trace-back written to the serial port in case of such a lock-up? Or might a watchdog+trace-back help to find the cause of this?
Resetting the receiver helps - the sender seems to be really stable.

Thank you,

Marc

MarcRWetzel, 2018-04-02:
I just found out, that the DI0 pin was not used (as it was not set to INPUT in the library) 
I am retesting now. I will run it for some hours and come back later :-)

brianmwebb, 2019-03-10:
pinMode(dio0, INPUT); in Setup