sandeepmistry/arduino-LoRa #131: problem compiling LoRaSender example :`SPISettings’ does not name a type`

yasharne, 2018-04-04:
Hi
When I want to compile LoRaSender example I get the following error:
```
In file included from LoRaSender.ino:2:0:
/home/yashar/sketchbook/libraries/LoRa/LoRa.h:84:3: error: ‘SPISettings’ does not name a type
   SPISettings _spiSettings;
```
How can I fix it?

morganrallen, 2018-04-05:
What board are you targeting?

yasharne, 2018-04-05:
arduino uno and nano

sandeepmistry, 2018-04-08:
What about Arduino IDE and AVR core version?

My suggestion is to update to the latest Arduino IDE, very old versions did not support SPISettings in the SPI library.

yasharne, 2018-04-09:
Thanks a lot for your help.
That solved my problem.
I was using arduino IDE downloaded via ubuntu package manager which is a very old version

closed 2018-04-28 by sandeepmistry