sandeepmistry/arduino-LoRa #133: We need an issue template

morganrallen, 2018-04-06:
To counter the growing onslaught of "LoRa failed to start" issues I believe we need an issue template.
The template should contain....

Devboard being used. (provide link)
Non-commercial board? Provide link & detailed pin description
Pin configuration
Did you check pins/`setPins` match?

@sandeepmistry any other thoughts?

MarcRWetzel, 2018-04-06:
In my opinion you are half-way right - a new user (as me, just a couple of days ago) would also just need a FAQ, to look thru, maybe some examples for known boards..., typical mistakes you normally make at the beginning

Just my 2ct

sandeepmistry, 2018-04-08:
@morganrallen sounds good!

As @MarcRWetzel we could have a link to a FAQ or trouble shooting guide as well.

Another item is, are you using a level converter, and did you try to set `setSPIFrequency` if it doesn't support high frequencies.

Minimal example sketch(es) to reproduce issues would be good for more non-trivial issues.

morganrallen, 2019-07-10:
I've implemented this now, just points out the FAQ and `setPins` API, hopefully this will knock down some of those really basic issues.

closed 2019-07-10 by morganrallen