sandeepmistry/arduino-LoRa #134: Lora not sending packets shortly after previous send?

hb9eue, 2018-04-07:
Hello

I'm using this library to transmit NMEA GPS Sequences to sort of build a tracker. Of course not all sequences as the link would be too slow, but the $GPRMC and $GPGGA sequences which are most usefull. The attached GPS Module is sending the sequences once in a second, but the GPRMC and GPGGA sequence follow each other. Now I have the problem, that the GPGGA which I send via Lora just after I sent the GPRMC is not being sent. But the return values all are OK.

Code Snipplet:

From the main loop:

      if (gpssentence.startsWith("$GPGGA") || gpssentence.startsWith("$GPRMC")) {
         Serial.print(gpssentence);
         sendLora(gpssentence);
      }

int sendLora(String payload) {
  digitalWrite(LED_BUILTIN, HIGH);
  int length;
  if (LoRa.beginPacket()) {
    Serial.println("LoraBegin OK");
  } else {
    Serial.println("LoraBegin Failed");
  }
  length = LoRa.print(payload);
  Serial.printf("Wrote %d Bytes\n", length);
  if (LoRa.endPacket()) {
    Serial.println("Lora End OK");
  } else {
    Serial.println("LoraEnd Fail");
  }
  digitalWrite(LED_BUILTIN, LOW);
}

I see and can confirm, that LoRa.print is being called for BOTH sequences, RMC and GGA, but as the two calls are in very short time one after the other only the first one (RMC) makes it to the receiver. Strangely, on the second call all return values are absolutely OK, so there is no was to catch an arror and retry to send the packet.

If I artificially delay the sending of the second sentence by 200ms or so, it is getting through (but then I start missing sentences on the serial GPS side while waiting).

Did I stumble over a bug, or is this expected behavior?

morganrallen, 2018-04-07:
I've also observed behavior like this when sending large packets back to back (GPS data in fact). Is `length` being set as expected?

sandeepmistry, 2018-04-28:
Closing this for no, due to lack of activity.

closed 2018-04-28 by sandeepmistry

davidstoneham, 2019-09-30:
I have also encountered this issue. I'm sending large packets (100 bytes) back to back and eventually it stops actually sending the packets without any errors or hanging. 
Even if i go back to sending small packets after this it still does not actually send them unless i reboot the device.