sandeepmistry/arduino-LoRa #135: Follow-up question with using ATtiny85

3Dtj, 2018-04-09:
Sandeep,

Following up on @kadellabs' question (#128), I've compiled the following sketch to an ATtiny85 with the hope of illuminating an LED after receiving a signal from an arduino LoRa using the LoRa Duplex sketch .  While I did get the sketch to compile to the ATtiny85 that is connected to a RFM95 LoRa module, I'm unable to detect the signal.  Appreciate any suggestions. 

3Dtj

`*/
#include <SPI.h>              // include libraries
#include <LoRa.h>
  
const int receiveLED = 4;
const int csPin = 3;          // LoRa radio chip select
const int resetPin = 5;       // LoRa radio reset

byte localAddress = 0xFF;     // address of this device
byte destination = 0xBB;      // destination to send to
byte syncWord = 0xB4;        // sync word (network ID)
byte spreadingFactor = 8;    // spreading factor (6-12)

void setup () {
  LoRa.setPins(csPin, resetPin);  //set CS, reset, IRQ pin
  LoRa.begin(900E6);              // initialize ratio at 900 MHz  
  LoRa.setSyncWord(syncWord);
  LoRa.setSpreadingFactor(spreadingFactor);
  LoRa.setTimeout(10);                  // set Stream timeout 10 ms
  pinMode(receiveLED, OUTPUT); 
  }
void loop (){
onReceive(LoRa.parsePacket());
}
void onReceive(int packetSize) {
  if (packetSize == 0) return;          // if there's no packet, return
  digitalWrite(receiveLED, HIGH);       // turn on the receive LED
  int recipient = LoRa.read();          // recipient address
  String incoming = LoRa.readString();  //payload of packet
  digitalWrite(receiveLED, LOW);      // turn off receive LED
}`

sandeepmistry, 2018-04-28:
Hi @3Dtj,

I would just you check the power supply.

3Dtj, 2018-04-29:
Will do!   Thanks

3Dtj, 2018-05-06:
I've confirmed that 3.3V is reaching the RFM95 chip (connected voltmeter to the GPIO 3 pin and to ground).

Since I'm compiling the sketch from the Uno to the ATtiny, do I need to compile the libary code also?   With the ATtiny unable to reach back into the library folder on the PC, I'm thinking the sketch must also include the code for the SPI and LoRa libraries. 

 I've gotten the sketch to work for a RFM95 connected to my Uno/PC and the only change that I make when compiling to the ATtiny is renumbering the pins for the LED, csPin, and resetPin to correspond to the ATtiny.  Very odd.

3Dtj, 2018-05-16:
So I got the problem resolved after insight from cle ( https://github.com/ClemensRiederer/TinyLoRa-BME280_v1.1 ).  Not sure exactly why but cle connects the ATtiny's PB1 (MISO/DO) to the RFM95's MOSI terminal.  While this is opposite of what is done using the Arduino with Sandeep's LoRa duplex sketch, it worked out. 

Appreciate any insight on why the reversing the MISO and MOSI resolves the issue.

sandeepmistry, 2018-07-30:
@3Dtj can this be closed?

3Dtj, 2018-07-31:
Yes please.  Thanks Sandeep.

Travis

closed 2018-08-06 by sandeepmistry