sandeepmistry/arduino-LoRa #143: RAK811 problems

sabas1080, 2018-04-23:
Hi

I try to port the RAK811 to the Arduino Core, but I have some problems wanting to use the Lora module with this and other libraries Lora, I do not know how the pins work

RADIO_RF_CRX_RX PB_6 // CRF3 (ANT_SWITCH_TX_BOOST))

RADIO_RF_CBT_HF PB_7 // CRF2 HF (ANT_SWITCH_TX_RFO) 

RADIO_RF_CTX_PA 

The SPI working OK, but I can not do data transmission o receive, any help is welcome

I am using Arduino_Core_STM32 with this pullrequest

https://github.com/stm32duino/Arduino_Core_STM32/pull/229

Schematic
[RAK811 SCH.pdf](https://github.com/sandeepmistry/arduino-LoRa/files/1939527/RAK811.SCH.pdf)

Thanks

sandeepmistry, 2018-04-28:
@sabas1080 unfortunately I don't have any advice for you at this time. Let's leave this open for now though ...

wero1414, 2018-05-28:
It seems to be a problem with the way you configure the RF switch, its important to activate V3 and deactivate V2 and V1 to receive and, in this part im not sure the way to transmit, i think that you need to activate V2 and V1 and deactivate V3. 

[image]

ayushsharma82, 2018-06-30:
@sabas1080  Any Success ? I was looking to use RAK811 standalone too.

halukmy, 2018-06-30:
@sabas1080  speak about class-c lorawan

sabas1080, 2018-06-30:
@halukmy  it is uncomfortable in each issue to see comments on class C, it is a private code for a private company, I will not share, I hope you understand, thanks

sabas1080, 2018-06-30:
@ayushsharma82 the status is still the same, if you want to help is welcome in my repository https://github.com/sabas1080/Arduino_Core_STM32

closed 2025-09-01 by sabas1080