sandeepmistry/arduino-LoRa #144: ESP8266-12E with LoRa RA-01

sonukushwah221, 2018-04-26:
Hello Sandeep 

I am using MQTT with ESP8266-12 and lora Module the working like this.

Mqtt--> ESP8266--> LoRa  so it's throw the error 

Soft WDT reset

ctx: cont 
sp: 3fff02a0 end: 3fff05e0 offset: 01b0

>>>stack>>>
3fff0450:  00000000 3ffef2e0 3ffef2dc 402029db  
3fff0460:  007a1200 00000001 3ffef2dc 40106895  
3fff0470:  00000000 3ffef2e0 3ffef2dc 40202ae4  
3fff0480:  00000083 3ffef2e0 3ffef2dc 40204f1c  
3fff0490:  3ffef1e8 00000008 3ffef2e0 40202b27  
3fff04a0:  3ffef1e8 00000008 3ffef2e0 40202ba3  
3fff04b0:  3ffe89a8 3fff04d0 3ffef2e0 4020251a  
3fff04c0:  00000001 00000001 3ffef1f8 40202569  
3fff04d0:  3fff1b54 0000000f 00000001 3fff1ddc  
3fff04e0:  0000000f 00000001 3fff0510 40100690  
3fff04f0:  00000000 00000000 3fff1a74 40204f12  
3fff0500:  3fff1a74 3ffef238 3fff1a74 40204f4e  
3fff0510:  00000000 00000000 00000000 4020834c  
3fff0520:  3fff1a74 3ffef238 3ffef1f8 40204fd5  
3fff0530:  3fff1d94 0000000f 00000007 40203c70  
3fff0540:  3ffef238 00000001 3ffef1f8 4020427c  
3fff0550:  00000001 00000000 402043c4 0000000f  
3fff0560:  00000000 3fff1b1c 3ffef1f8 3ffef5b8  
3fff0570:  00000001 3ffef21c 3ffef1f8 402051ec  
3fff0580:  3ffe8cb0 00000000 00001388 40208ddc  
3fff0590:  00000000 3fff1b1c 00000000 40207cc9  
3fff05a0:  3fffdad0 00000000 3ffef5b0 402025a5  
3fff05b0:  00000000 00000000 00000001 40208b5d  
3fff05c0:  3fffdad0 00000000 3ffef5b0 40208b88  
3fff05d0:  feefeffe feefeffe 3ffef5c0 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 

====================================
Decodec
====================================

Decoding 24 results
0x402029db: SPIClass::beginTransaction(SPISettings) at C:\Users\kapil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\SPI/SPI.cpp line 512
0x40106895: __digitalWrite at C:\Users\kapil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/core_esp8266_wiring_digital.c line 84
0x40202ae4: LoRaClass::singleTransfer(unsigned char, unsigned char) at C:\Users\kapil\Documents\Arduino\libraries\LoRa\src/LoRa.cpp line 518
0x40204f1c: FunctionRequestHandler::handle(ESP8266WebServer&, HTTPMethod, String) at C:\Users\kapil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 214
0x40202b27: LoRaClass::readRegister(unsigned char) at C:\Users\kapil\Documents\Arduino\libraries\LoRa\src/LoRa.cpp line 518
0x40202ba3: LoRaClass::endPacket() at C:\Users\kapil\Documents\Arduino\libraries\LoRa\src/LoRa.cpp line 518
0x4020251a: sendMessage(String) at C:\Users\kapil\Documents\Arduino\Lora_New_MQTT/Lora_New_MQTT.ino line 85
0x40202569: operator() at C:\Users\kapil\Documents\Arduino\Lora_New_MQTT/Lora_New_MQTT.ino line 66
:  (inlined by) _M_invoke at c:\users\kapil\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2071
0x40100690: free at C:\Users\kapil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1737
0x40204f12: std::function ::operator()() const at C:\Users\kapil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 214
0x40204f4e: FunctionRequestHandler::handle(ESP8266WebServer&, HTTPMethod, String) at C:\Users\kapil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 214
0x4020834c: String::String(String const&) at C:\Users\kapil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/WString.cpp line 545
0x40204fd5: ESP8266WebServer::_handleRequest() at C:\Users\kapil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 214
0x40203c70: WiFiClient::~WiFiClient() at C:\Users\kapil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 329
0x4020427c: WiFiServer::available(unsigned char*) at C:\Users\kapil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src/WiFiServer.cpp line 106
0x402043c4: FunctionRequestHandler::canHandle(HTTPMethod, String) at C:\Users\kapil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 214
0x402051ec: ESP8266WebServer::handleClient() at C:\Users\kapil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 214
0x40208ddc: std::_Function_handler ::_M_invoke(std::_Any_data const&) at c:\users\kapil\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069
0x40207cc9: HardwareSerial::available() at C:\Users\kapil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/HardwareSerial.cpp line 174
0x402025a5: loop at C:\Users\kapil\Documents\Arduino\Lora_New_MQTT/Lora_New_MQTT.ino line 174
0x40208b5d: esp_schedule at C:\Users\kapil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/core_esp8266_main.cpp line 57
0x40208b88: loop_wrapper at C:\Users\kapil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/core_esp8266_main.cpp line 57
0x40100710: cont_norm at C:\Users\kapil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/cont.S line 109

Here I show the Error with decode

sandeepmistry, 2018-04-28:
@sonukushwah221 does the issue still occur if you send the same data, but remove the MQTT side?

sonukushwah221, 2018-04-30:
@sandeepmistry  if we remove the MQTT code when data is sent properly.