sandeepmistry/arduino-LoRa #146: LoRa receiver code not working

pritibakale, 2018-05-08:
Hello,

I am using examples from this LoRa library. When I used code LoRaSender then it is working fine and sends data. But to receive the message when I used LoRaReceiver code then,  it is not at all receiving anything.

Can anyone please suggest, what could be the issue?

sandeepmistry, 2018-05-27:
@pritibakale are you passing the correct frequency into `begin` for the hardware you are using. Same applies to antenna setup

Depending on what h/w you are using, you might need to use the [setTxPower](https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#tx-power) API to use PA BOOST for power.

sandeepmistry, 2018-07-30:
Closing for now due to lack of feedback.

closed 2018-07-30 by sandeepmistry