sandeepmistry/arduino-LoRa #147: Micro-meetup at BAMF

morganrallen, 2018-05-16:
Hi @sandeepmistry and other contributors! After seeing you (Sandeep) will be at BAMF on the Twitters I thought it would be great to meet up. I'll be working the event on Friday/Sunday in the Experiments in Space area and generally hanging out Sat (Tindie meetup) / Sun (Supply Frame meetup) evenings.

Hope to see all the LoRa hackers around!

sandeepmistry, 2018-05-17:
@morganrallen it would be awesome to meet up!

I'm not sure what my schedule is at this point, but you should be able to find me at the Arduino/Microchip booth in zone 2 for the most part. Let me see if I can head to the meetups you suggested as well ...

sandeepmistry, 2018-05-27:
@morganrallen and I had a chance to briefly meet in person. Thanks for finding me!

closed 2018-05-27 by sandeepmistry