sandeepmistry/arduino-LoRa #151: board issue...

JackHsuan, 2018-05-30:
hello,
I wanna know that the board:linkit7697 is suitable for this lora library?

I use the board and when I complie it ,I get an error:
/Users/jackhsuan/Documents/Arduino2/libraries/arduino-LoRa-master/src/LoRa.cpp: In member function 'void LoRaClass::onReceive(void (*)(int))':
/Users/jackhsuan/Documents/Arduino2/libraries/arduino-LoRa-master/src/LoRa.cpp:329:48: error: 'digitalPinToInterrupt' was not declared in this scope
     attachInterrupt(digitalPinToInterrupt(_dio0), LoRaClass::onDio0Rise, RISING);
                                                ^
/Users/jackhsuan/Documents/Arduino2/libraries/arduino-LoRa-master/src/LoRa.cpp:331:48: error: 'digitalPinToInterrupt' was not declared in this scope
     detachInterrupt(digitalPinToInterrupt(_dio0));
                                                ^
exit status 1
board LinkIt 7697 compile error。

sandeepmistry, 2018-07-30:
@JackHsuan the LinkIt core is missing  `digitalPinToInterrupt` Arduino API, I suggest you ask the LinkIt core maintainers to add it.

grafeoy, 2019-06-13:
Reduce the library version to 0.0.1

dontsovcmc, 2020-05-19:
I can't understand..  I have compile error with Atmega32u4 board. 

```
[env:feather32u4]
platform = atmelavr
board = feather32u4
framework = arduino
```

closed 2021-06-11 by IoTThinks