sandeepmistry/arduino-LoRa #155: Encrypt support

halukmy, 2018-06-07:
When you will add secure message support ?

sabas1080, 2018-06-07:
Hi

you can use an encryption library

example https://github.com/rweather/arduinolibs

halukmy, 2018-06-07:
@sabas1080  is it makes secure communication?

wero1414, 2018-06-07:
Secure as you keep your keys hidden, if you need really secure communication you should try LoRaWAN

sandeepmistry, 2018-07-30:
No plans to add encryption to this library.

closed 2018-07-30 by sandeepmistry

alaeddineth, 2020-06-03:
> No plans to add encryption to this library.

I'm doing a  projet to secure this library but I found a weird problem 
our job is actually:
Sender:
send a message that contains
* Information encrypts with aes128 with session key
* hashing with sha256
* digital signature obtained with the asymmetric curve crypto algorithm
* the session key
reciever:
* will decrypt the data with the session key (confidentiality)
* decrypt the digital signature with the shared secret key and obtain the hash to ensure authentication
* will hash the data and compare it with the hash received to verify the integrity of the data

we tested all its functions one by one, and we applied them on the lora sender, the code is successfully compiled but the output: contains nothing even the display in serial port and the message was not transmitted
I did not really understand what the problem is with insufficient memory or the arduino takes too long to calculate data or what
here is the code
and the output in screenshot
[image]
[image]
[image]

halukmy, 2023-01-22:
@alaeddineth can you make a sample rf tx for this ?