sandeepmistry/arduino-LoRa #156: Dragino LoRa Shield + Ethernet Shield (W5100)

TupaNegreiros, 2018-06-09:
The documentation is not clear (even Dragino wiki).
In order to use "setPins", do I need to change jumpers SV2/SV3/SV4 positions?

I'm trying to use Dragino LoRa Shield together with Ethernet Shield (W5100), both shields use pin 10 as SS. Seems impossible to change SS on Ethernet Shield, and unclear how to change it on LoRa.

sandeepmistry, 2018-07-30:
> In order to use "setPins", do I need to change jumpers SV2/SV3/SV4 positions?

Yes, then pass the pins to `LoRa.setPins(...)`.

closed 2018-07-30 by sandeepmistry

fer-g3, 2020-09-04:
I can´t make it work, I have an Arduino Uno + Shield Ethernet + Shield LoRa 915Mhz. I already change the pins with setPins function but it always return LoRa init failed. Check your connections.

JuandreG, 2022-04-09:
Could you get this setup working? I'm having the same issue.