sandeepmistry/arduino-LoRa #157: Implementing SX127x errata for 500KHz bandwidth

vasimv, 2018-06-11:
The Semtech did make an errata for the SX127x chips, stating it requres a bit manual tuning when bandwidth set to 500KHz. I've tried to implement it and it shows few db's increase on receiver.

Errata pdf: https://www.semtech.com/uploads/documents/sx1276_77_78-errata.pdf

[patch-errata.txt](https://github.com/sandeepmistry/arduino-LoRa/files/2090620/patch-errata.txt)

torntrousers, 2018-06-12:
Thats interesting, a few db's is quite a significant change. I'll give it a try too. 
Doesn't your update need to take account of the frequency - the errata says use value 0x64 when  the frequency is above 862MHz?

vasimv, 2018-06-12:
Oops, sorry, should say. I'm using 433MHz and you're right.

sandeepmistry, 2018-07-30:
@vasimv please submit a pull request for this.

IoTThinks, 2021-06-11:
Already fixed.
https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#signal-bandwidth

closed 2021-06-11 by IoTThinks