sandeepmistry/arduino-LoRa #158: How to set AppsKey and NwsKey or frequency channel?

TomHyhlik, 2018-06-12:
I did not find ability to set these parameters. Why it's not involved in this library? How to implement it?

TomHyhlik, 2018-06-13:
Thank you very much 👍

halukmy, 2018-06-13:
@wero1414  can you share aes eencryypted message sample

wero1414, 2018-06-14:
@halukmy this library explains how to encrypt your message you need a key in the tx and the rx https://github.com/DavyLandman/AESLib .

This example is in the readme

 `Serial.begin(57600);
uint8_t key[] = {0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15};
char data[] = "0123456789012345"; //16 chars == 16 bytes
aes128_enc_single(key, data);
Serial.print("encrypted:");
Serial.println(data);
aes128_dec_single(key, data);
Serial.print("decrypted:");
Serial.println(data);`

wero1414, 2018-06-14:
Closing cause doubt solved, and provide an aes example

closed 2018-06-14 by wero1414

halukmy, 2018-06-19:
@wero1414  i got 2 question for you
t
hanks for share first..

can we secure our message from sniffers?

and second is, do you have got Class-C for LoRaWAN in Lmic code?

thanks bro