sandeepmistry/arduino-LoRa #159: Can't receive at 62.5kHz BW setting

Yury-MonZon, 2018-06-13:
Everything is fine (2km in the city) when I use these settings:
` LoRa.setTxPower(POWER);  //20 max power
  LoRa.setSpreadingFactor(12);  //12 for max range
  LoRa.setSignalBandwidth(125E3); //Supported values are 7.8E3, 10.4E3, 15.6E3, 20.8E3, 31.25E3, 41.7E3, 62.5E3, 125E3=default, and 250E3.
  LoRa.setCodingRate4(8);   //Supported values are between 5 and 8, these correspond to coding rates of 4/5 and 4/8. The coding rate numerator is fixed at 4.
  LoRa.enableCrc();`

But when I switch to 62.5E3, I receive nothing at all. I've tried 10.4E3 and 20.8E3, - those settings worked fine.

halukmy, 2018-06-13:
@Yury-MonZon  what is your antenna?

Yury-MonZon, 2018-06-13:
Stock microstrip pcb-sticker antenna. Probably it is not the best, but it works on other BW settings. But with 62.5k I can't get it to work even 10 cm apart.

halukmy, 2018-06-13:
how you send message over 2km with that antenna? @Yury-MonZon

Yury-MonZon, 2018-06-14:
2km in the city, so lots of reflections etc. I don't know, it just works. My settings in the first post. For the power I used dontsovcmc patch for 20dbm support (https://github.com/sandeepmistry/arduino-LoRa/pull/153/commits).

halukmy, 2018-06-16:
@Yury-MonZon  what are you using antenna? can yous hare on aliexpress?

Yury-MonZon, 2018-06-17:
@halukmy Please, check my post from June 13. I have the antenna linked there.

halukmy, 2018-06-17:
@Yury-MonZon  thanks i saw it before but is'nt it so little? for example gateway antennas like https://www.aliexpress.com/item/868MHz-high-gain-12dBi-fiberglass-antenna-868MHz-outdoor-roof-antenna/32313781076.html that much big

your two nodes are same antenna ? is that possible 2km?

thanks

Yury-MonZon, 2018-06-17:
@halukmy Yes, I do have only those antennas and I'm using 433MHz band, not 866. It just works.

halukmy, 2018-06-17:
wow amazing.. @Yury-MonZon  have you ever work with LoRaWAN?

Yury-MonZon, 2018-06-18:
Nope, I'm using it without the LoRaWAN stack, like ordinary modems. Why?

halukmy, 2018-06-18:
@Yury-MonZon just wonder,

so how you secure communication from sniffers?

Yury-MonZon, 2018-06-18:
Encryption. But this is not relevant to this issue.

halukmy, 2018-06-19:
@Yury-MonZon yeah i know but how you planning to secure communication with this lora-wan repo?

Yury-MonZon, 2018-06-20:
I'm not planning to use LoRa WAN stack yet. At the moment I'm focused on using LoRa as radio modem.

halukmy, 2018-06-20:
Super cool lets work together about secure our message from sniffers @YuryMonZon

Yury-MonZon, 2018-06-20:
@halukmy I will have some encryption but only after this library would be improved ie temperature drift etc...

halukmy, 2018-06-20:
I will help you temp stuff and you help on crypt

Yury-MonZon, 2018-06-21:
@halukmy Sure! Let me know when you'll have it ready.

halukmy, 2018-06-21:
@Yury-MonZon mail me mate halukmy@gmail.com

loraraspi91, 2020-05-26:
I have the same issue with RFM95 module. working fine at 125 khz and 250 khz. But I got CRC error at 62.5 Khz. And no reception at below  bandwith. I tried with various antenna, got the same. 
Maybe a limitation of the RFM95 ?

AaronK9, 2020-05-26:
It’s possible there is enough error in the center frequencies between the two radios that one doesn’t hear the other. This happens at lower bandwidths. I believe that the radios can tolerate ~25% (of the BW) error before this happens. The smaller the BW, the more this is a problem. 

Solution: Send a message at a higher bandwidth and check the frequency error (function in this library). You must then tune that offset (add or subtract), on the radio you measured, to get the two radios on the same actual frequency. After you’ve adjusted your set frequency in your sketch, try a smaller BW. 
Note: temperature may have an effect on this offset due to the lower quality components in the LoRa module. Hope this helps!

loraraspi91, 2020-05-27:
Thanks you for your reply,

I did what you mentioned.

at high bandwidth even with an error of 4000 I can receive data with no issue. So I corrected the frequency to get an error less than 15.  Finally I change to lower bandwidth (62.5) and I still have CRC error and most of the time no reception. 

Even with a frequency Error of 37 (with 62.5€3 BW) I still have CRC error and sometime I can receive correct message with 181 as frequency error. Aslo, when I do not receive anything I do not know the error. 

For me, it is like the module do not receive correctly the message (maybe the BW is not correct/consistent between them). Maybe a small variation in the BW and they do not receive. I do not understand since I'm using for both RFM95 module.

IoTThinks, 2020-05-27:
Should use same type of antennas in both sides?

A bad antenna may create more frequency error.

loraraspi91, 2020-05-27:
Finally I found out the issue:
with SF12 the minimum BW is 125Khz.  if you select below like 62.5 khz I got CRC issues or not receiving
with SF11 the minimun BW is 31250 khz. So with SF11, I can have a BW of 62.5 khz.
with SF10 the BW 15600 is working (not try below). 

So there is a link between the SF and the minimum BW allowed : but I do not understand why ?

closed 2021-07-04 by IoTThinks

Kongduino, 2021-07-04:
There is a link between the SF/BW combination and the maximum payload you can send. If your messages are garbled, they're too long. At SF 12/BW 6, the max payload is around 51 bytes. Above that and you will have errors. I'm doing distance tests at 468 MHz, SF 12, BW 6, and they work – because the packets are very small. Anything over 50+ (it's not an exact science, I got away with 55 and 59 I think), you should switch to something else. If you're trying to achieve range, SF 12 & BW 125 is probably your best bet.