sandeepmistry/arduino-LoRa #160: Issue if a new packet arrives before the first is complete?

dwaynez, 2018-06-14:
I have based my code on the duplex example. At random times, the program seems to be going into an infinite loop. My code is below. I think somehow the function available is returning a negative value. Is it possible for the register REG_RX_NB_BYTES to change value while in the middle of processing a read packet?

    if (LoRa.parsePacket() > 0) {
        int incoming_index = 0;
        byte incoming[s_msg_length];
        while (LoRa.available()) {
            if (incoming_index < s_msg_length) {
                incoming[incoming_index] = (byte)LoRa.read();
            } else {
                byte x = (byte)LoRa.read(); // Ignore the message beyond the buffer length
            }
            incoming_index++;
        }

morganrallen, 2018-07-29:
First off `available` *shouldn't* be able to return a negative despite it being an `int`. It simply returns `readRegister` which is `uint8_t`.

Can you put together a minimal example the displays this behavior? Are you doing anything with explicit/implicit headers? **Operations of Operation** on page 35 in the [datasheet](http://www.mouser.com/ds/2/761/sx1276-944191.pdf) makes note on `REG_RX_NB_BYTES` in explicit vs implicit, maybe of help.

dwaynez, 2018-07-30:
I seem to have resolved my problems. I don't know why yet, but the problems all seem to be related to me trying to monitor the voltage of the power supply (using s voltage divider). Every time I try to do this, I get all kinds of weird errors on the LORA. As soon as I stop doing this, the system is rock solid. Any idea of why this would be such a problem to monitor input voltage?

morganrallen, 2018-07-30:
*hand wavy* Your monitoring could be causing power dips, adding a largish capacitor might help.

closed 2018-07-30 by morganrallen