sandeepmistry/arduino-LoRa #162: Xx-tea cryptology algorythm for lora-arduino

halukmy, 2018-06-19:
@dontsovcmc 

i happy to see you starred that repo, so you think xx-tea-iot for lora-arduino secure?

dontsovcmc, 2018-06-19:
I know, that XTEA algorithm is easy way to secure your data for embedded devices. As you know secure for one project isn't secure for another. I don't read abstracts about serurity in IoT.

info: https://en.wikipedia.org/wiki/XTEA
online: https://sanstv.ru/tools/crypt/alg-xtea

halukmy, 2018-06-19:
thanks for share,

on our repo lets think about secure communication,

first i secure my message is "on1" to "@3W1£e" it loraX message

my key is "loraX"

sending one node works "as gateway" to another node this crypted message

on node side,  key is "loraX"

than uncrypt message and turn on light for example..

if someone sniff my lora message and send this message "@3W1£e" , node will act again as me..

so how i can secure it from any copier or sniffer my lora node ?

dontsovcmc, 2018-06-19:
I don't know ISO about cryptography, but I think you can read this article:
1. HTTP Digest auth solve your problem https://en.wikipedia.org/wiki/Digest_access_authentication
2. simple way: use random number and increase it every message
3. Google solving this: https://en.wikipedia.org/wiki/Replay_attack

dontsovcmc, 2018-06-19:
example message fields:
| device id (uint32) | random (uint32) | message id from 0 (uint32) | body |

When you receive message you should increase message id.
When you receive message with id != id +1 - ignore it.

sandeepmistry, 2018-07-30:
@halukmy please stop opening duplicate issues on this repo.

Duplicate of #155 and #161.

closed 2018-07-30 by sandeepmistry