sandeepmistry/arduino-LoRa #164: LoRa Sender Starting LoRa failed!

mavericks123, 2018-06-25:
Hello,

I have a problem with sending and receiving data. 

I followed steps like on your post, but serial monitor says:

LoRa Sender
Starting LoRa failed!

A code:

`#include <SPI.h>
#include <LoRa.h>

int counter = 0;

void setup() {
  Serial.begin(9600);
  while (!Serial);
  LoRa.begin(433E6);

  Serial.println("LoRa Sender");

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;

  delay(5000);
}`

[image]
[image]
[image]

Yury-MonZon, 2018-06-28:
Looks like LoRa is not responding. Check the cabling, power etc.

morganrallen, 2018-07-29:
@mavericks123 You need to ensure the pins are all mapped correctly.

SPI pins `MISO`, `MOSI`, `SCK`, `SS` need to be connected to the Uno's SPI pins
```
10 (SS),
11 (MOSI),
12 (MISO),
13 (SCK)
```

Then you need to ensure your `RESET` and `DIO0` pins match *or* call [setPins](https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-pins)

It's hard to tell from the pictures but it does seem you're setup fairly close but it's worth looking over everything closely.

sandeepmistry, 2018-08-19:
Closing this for now due to lack of activity.

closed 2018-08-19 by sandeepmistry