sandeepmistry/arduino-LoRa #166: LoRa.begin() works. But LoRa.beginPacket,LoRa.endPacket() and LoRa.print() not working

pritirathi14, 2018-06-28:
Hi Sandeep,

I am interfacing LoRa sx1276 with Arduino MEGA. The connections are all good. The setup code of Arduino is working fine.

void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
delay(500);
while (!Serial);

  Serial.println("LoRa Sender");

  if (!LoRa.begin(868E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

=======================

But below lines are not working

LoRa.beginPacket();
  LoRa.print(ktc.readCelsius());
  LoRa.endPacket();

execution seems to be halted at the LoRa.beginPacket() line.

Yury-MonZon, 2018-06-28:
Try 
```
      LoRa.beginPacket();
      LoRa.print("Test");
      LoRa.endPacket();
```

pritirathi14, 2018-06-29:
Its not working.
Even if I try 

LoRa.beginPacket();
      LoRa.print("Test");
      LoRa.endPacket();
Serial.println("Hello");

the serial print is also not working. Seems control is stuck at the LoRa.beginPacket() line.

Yury-MonZon, 2018-06-29:
I'm afraid that I have to see the whole code in order to crack it. Drop it to pastebin and link here.

sandeepmistry, 2018-07-30:
@pritirathi14 please see 
https://github.com/sandeepmistry/arduino-LoRa/issues/166#issuecomment-401247819, also you might want to try with an external 3.3V power supply.

sandeepmistry, 2018-09-04:
Closing this for now due to lack of activity.

closed 2018-09-04 by sandeepmistry

taherrera, 2019-04-26:
I am using the MKRWAN1300. For me it hanged when the buffer was "big" (8 bytes or more). I noticed the SPI frequency was 250kHz, kinda low.. anyway when I increased the frequency it never worked, so further decreasing the SPI frequency to 200kHz made it work for me..

An other problem I had, and I hope it helps somebody else: it does not work without the external antenna !! 

ps: MKRWAN1300 sucks, what a waste of money..

bcaceres96, 2024-10-16:
Has anyone managed to solve the error?