sandeepmistry/arduino-LoRa #168: Esp32 SX1278 compability

craxen, 2018-06-29:
Hello, im trying to use your magnificient library with my ESP32 \SX1278 module, this does not work, could you perhaps fix it ?

Im trying the nolora sketches, and i get complaints about avr\progmem missing.
I would love to be able to control all my rc-switches.

Other library functions seem to work with my ttgo lora besides the remotec.

Marten

morganrallen, 2018-07-26:
This sounds like you're not setting your board correctly in the Arduino interface but need more info to be sure.

craxen, 2018-07-26:
Hello, ok, but all your other examples compile and work on the same board,
hmm, im going to try it again and give you some debug info.
Thanks

morganrallen, 2018-07-29:
> Im trying the nolora sketches, and i get complaints about avr\progmem missing.
I would love to be able to control all my rc-switches.

What are the "nolora sketches" ?

craxen, 2018-07-29:
Hello, i refer to the examples for remote control named noloraxxxx in your
examples folder

sandeepmistry, 2018-07-30:
There is no `noloraxxxx` example in this repo: https://github.com/sandeepmistry/arduino-LoRa/tree/master/examples

closed 2018-07-30 by sandeepmistry