sandeepmistry/arduino-LoRa #169: Spreading Factor settings with PJON

jwktje, 2018-07-03:
I'm using arduino-LoRa with PJON. It works great when using Spreading Factor 7, but when I change it to anything else it stops working. I'm communicating between two arduino's using the RFM95W. 

Is there anything else I need to configure for other spreading factors to work? I want to increase the range I get and I wanted to experiment with a higher SF but can't get it to work. 

Any help at all is really appreciated!

morganrallen, 2018-07-30:
Can you better describe "stops working"?

Does it fail on the sending side?
Do you simply not receive it?
Are you setting SF on both sides?

sandeepmistry, 2018-08-19:
Closing this for now due to lack of activity. Please re-open when requested info.

closed 2018-08-19 by sandeepmistry