sandeepmistry/arduino-LoRa #172: Arduino MKR WAN 1300 random() always returning 0

bravech, 2018-07-20:
I was trying to use the LoRa.random() function with the Arudino MKR WAN 1300, and every time it returns 0x00. I checked this by dumping the registers with a radio transmitting and with nothing transmitting, and the value didn't change. I noticed that in issue #59, @YannikW stated that register 0x1b is the raw RSSI, but that isn't the register used for measuring "Wideband RSSI", which is register 0x2c. 

Does anybody have a list of the registers and what they do? I couldn't find the document that @YannikW referenced.

sandeepmistry, 2018-08-06:
@bravech any progress on this?

Is PR #150 of any interest?

bravech, 2018-08-07:
I just worked around this by doing LoRa.parsePacket() before trying to
measure the RSSI.