sandeepmistry/arduino-LoRa #174: Connecting two LoRa Module to Arduino Mega

Ak4703, 2018-07-26:
Hi, 
I tried connecting two rfm95 modules to arduino mega. 
I have used the same SPI pins but different SS, RST and interrupt pins. 
I defined a new  Lora class and set the new SPI pins using the new lora class, but .begin() throws up an error?
Is there any way to achieve this?

morganrallen, 2018-07-26:
What's the error? Can you provide example code?

Ak4703, 2018-07-27:
`[code]
const char csPin = 10;          // LoRa radio chip select
const char resetPin = 9;       // LoRa radio reset
const char irqPin = 2;         // change for your board; must be a hardware interrupt pin

const char csPin2 = 8;          // LoRa radio chip select
const char resetPin2 = 7;       // LoRa radio reset
const char irqPin2 = 3;         // change for your board; must be a hardware interrupt pin
LoRaClass LoRaT;

void setup() {
  Serial.begin(9600);

  Serial.println("LoRa Receiver");
  LoRa.setPins(csPin, resetPin, irqPin);
  
  if (!LoRa.begin(866E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }

  LoRaT.setPins(csPin2, resetPin2, irqPin2);

  int v = LoRaT.begin(866E6);
  if (!v) {
    Serial.println(v);
    Serial.println("Starting LoRa Two failed!");
    while (1);
  }

[/code]`

sandeepmistry, 2018-08-06:
If you try just the second set of pins on the regular `LoRa` does it work?

Ak4703, 2018-08-08:
Yes.

sandeepmistry, 2018-08-19:
Have you tried an 3.3V external power supply?

What does `LoRa.dumpRegisters(Serial)` and `LoRaT.dumpRegisters(Serial)` output?

sandeepmistry, 2018-09-04:
Closing this for now due to lack of activity.

closed 2018-09-04 by sandeepmistry