sandeepmistry/arduino-LoRa #177: Add function to invert IQ signal register.

ricaun, 2018-07-31:
InvertIQ allows differentiating sender and receiver message. 

Usually, Gateways read messages with InvertIQ off and send messages with InvertIQ on, Nodes read messages with InvertIQ on and send messages with InvertIQ off.

This way a Gateway only reads messages from Nodes and never reads messages from other Gateway, and Node never reads messages from other Node.

I already test in my fork.
Only need to add this code.

```
#define REG_INVERTIQ             0x33

void LoRaClass::invertIQ(boolean invert)
{
  writeRegister(REG_INVERTIQ , invert == false ? 0x27 : 0x66);
}
```
Thanks

morganrallen, 2018-07-31:
It would be great to see this as a full PR. Should be able to just add it to `LoRa.h` and `API.md`

halukmy, 2018-08-07:
@ricaun  hey Luiz, how nodes understand other one is not node? or gateway?

for example , i am using this lib as gateway too, 

is 0x27 gateway  or 0x66 node?

ricaun, 2018-08-07:
@halukmy I create two examples, https://github.com/ricaun/arduino-LoRa/tree/master/examples

Basically Gateway send message with invertIQ enable and receive with invertIQ disabled.
Nodes sends message with invertIQ disabled and receive message with invertIQ enable.

LoRaWan use this arrangement, with invertIQ enable is possible to sniff downlink packages.

halukmy, 2018-08-14:
@ricaun  i been searching this been weeks... lorawan hard to implement but this repo quick and good as possible as..

the most matter thing is security and if we can disable sniffers from our messages, that would be awesooomeeee

halukmy, 2018-08-14:
@ricaun  one more question,

for example i got 1 gateway and 5 nodes

gateway is 0xAA

nodes like

0xA1
0xA2
0xA3
0xA4
0xA5

is it possible can we communicate 5 node and 1 gateway together and disable sniffers from messages?

sandeepmistry, 2018-08-19:
Closed via https://github.com/sandeepmistry/arduino-LoRa/pull/179.

closed 2018-08-19 by sandeepmistry