sandeepmistry/arduino-LoRa #180: Using LoRa Library in Energia for MSP430G2553

DeeLoRa, 2018-08-05:
I'm trying to compile LoRa library for MSP430 using Energia platform and I get the error SPISettings' does not name a type.

This is because the SPISettings is nowhere in the SPI library for Energia but is used in the above LoRa library.

My problem is how do I modify the LoRa library to add SPISettings or is there another method to solve this issue? 

``#include "Arduino.h"
#include <SPI.h>
#include <LoRa.h>
#include "Energia.h"
int counter = 0;

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Sender");

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;

  delay(5000);
}``

morganrallen, 2018-08-05:
I'm not familiar with `Energia` but it appears to be a fork of Arduino? If this is the case it would likely be out of the scope of this library. You might need to end up filing an issue with TI to get proper Arduino SPI compatibility.

sandeepmistry, 2018-08-06:
We could add support for board cores that don't have SPI transaction support if there is enough demand. There's an `SPI_HAS_TRANSACTION` macro available to detect it.

DeeLoRa, 2018-08-06:
Support on this would be great since LoRa is for Low powered IoT and MSP is also very low power oriented, so many will benefit from this.

sandeepmistry, 2018-08-06:
@DeeLoRa care to dive in an submit a pull request?

sandeep-singh4486, 2020-04-01:
Even i am facing his issue with energia with LORA with same SPI