sandeepmistry/arduino-LoRa #191: Transmission stops after a while

dwaynez, 2018-09-29:
I have built 2 LORA nodes with an Arduino Nano clone and an AI Thinker LORA module. First 2 worked great but the 3rd stops transmitting after a while. I modified endpacket() to only loop 5000 times before and indeed the TX_Done flag is never being set. I am having no problem communicating with the LORA using SPI, but the transmission never seems to be working. I dump registers before transmission begins and then once it starts but all appears normal, just that the TX_Done flag is never set.

The only way to recover is to power off the LORA and power back on again. Rebooting the Arduino does not help. There does not seem be anyway to have software to recover from the situation. I currently don't have anything connected to the RESET terminal of the LORA module. I was suspecting power as an issue, but I don't know why it would be an issue. All 3 units I have built use a 12V sealed battery and the LORA is feed directly from the 12V battery, through a buck converter to get down to 3.3V.

Any thoughts on what cause the TX DONE flag to never be set?

ricaun, 2018-10-03:
Hello,
I have this problem one time, sometimes the board send the package but never receive the TX flag and stay on a while forever. 
I guess sometimes the program clears the TX flag on the dio0 interrupt before the main program notice, I like to use onRecieve.
I put a timeout inside the while on the endPacket and 'fix' the bug.
Now I use onTxDone to check the TX flag interrupt, check my fork or the PR.
https://github.com/sandeepmistry/arduino-LoRa/pull/188

dwaynez, 2018-10-05:
I am still figuring that my problem is power related, but I don't know why. When I run from a 12V wall wart, all is well. When I run from a 12V SLA battery, it fails after a short period of time. I connected the 3.3V power supply to a cheapo USB oscilloscope and there certainly is much more variability in the voltage from the battery than from the wall wart, but still above the minimum for the chip (2.7V). I don't really understand why. Maybe there is a very short term spike down and this causes transmission to fail?

I did discover something interesting -- after transmission has failed (I use a loop counter in end_packet), the value of the REG_MODEM_CONFIG_1 has been changed from 0x72 to 0x12, which says that the bandwidth has been changed from 125kHz to 7.8kHz. I didn't change it. Why is the chip changing this value on its own?

ricaun, 2018-10-05:
Probably some power craziness...
Your Arduino Nano Clone is 3.3V?!
Are you using a StepDown regulator?! Try to change to LM1117 3.3V or 2 good AA battery.

It's possible to change the fuse of the chip to make work with less them 2.7V. 
Change the extended_fuses to 0xfe, this fuse set the BOD to VCC = 1.8V.
Check the fuse here: http://www.engbedded.com/fusecalc/

dwaynez, 2018-10-05:
The nano clone is 5v and I am using a variable buck converter to get 3.3V for the LORA.

I made a change to force the REG_MODEM_CONFIG_1 register to 0x72 in Being_packet and so far so good. I haven't seen anyone talk about this register changing dynamically so I am at a loss as to why it is occurring.

ricaun, 2018-10-05:
>The nano clone is 5v and I am using a variable buck converter to get 3.3V for the LORA.
>
Test to change to a 3.3 linear regulator. Are you using any logic level converter?

> I made a change to force the REG_MODEM_CONFIG_1 register to 0x72 in begin_packet and so far so good. I haven't seen anyone talk about this register changing dynamically so I am at a loss as to why it is occurring.
>
Something is really wrong, I don't know what could be.

dwaynez, 2018-10-05:
Yes, I am using a logic converter. Thinking that it might have been a bad component, I have swapped everything - buck converter, nano, logic leveler, LORA - same results, the only thing I didn't change was the battery. Weird I know.

dwaynez, 2018-10-11:
So after a lot of testing, this is what I have found:

If I run the system connected to a wall wart with 12V power -- no problems.

If I run the system connected to a 12V SLA battery then I get fluctuations in voltage, between 3 and 4 voltages (by buck is set to 3.6). These voltage fluctuations seem to cause problems with my AI Thinker RA-01 LORA module and the contents of the REG_MODEM_CONFIG_1 register is being changed to all kinds of crap. I have put in statements that do a reset of this register to 0x072 after I do a parsepacket (for reading) and after a beingpacket (for writing). The result is a solid communication between my Arduino Nano and my Raspberry Pi.

Just one other comment -- I am using a battery maintainer to charge the 12V SLA. I wonder if it is causing the voltage fluctuations, even when it is not plugged in? Just a thought, but I am not going to test further.

Back to my original issue -- setting the LORA to transmit and waiting for the transmit done IRQ to be set does not work because the REG_MODEM_CONFIG_1 was garbage (0x00, 0x12, 0x30 etc.). When it is properly set to 0x72, no issue.

closed 2018-10-11 by dwaynez