sandeepmistry/arduino-LoRa #192: Issue about Starting LoRa failed!

seyment, 2018-09-30:
Hi ,
     I have 433M dragino lora shield with rf98 chip And I also use arduino mega. I used lora sender and lora receiver in both sides and both gave me the same problem Starting LoRa failed. My code is:
#include <SPI.h>
#include <LoRa.h>

int counter = 0;

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Sender");

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;

  delay(5000);
}

Flavio-da-Brescia, 2018-11-28:
I have the same problem with Arduino Mega and LoRa RA-02
The same module and the same code work with Arduino UNO, but when I use the arduino MEGA it wont work
LORA------------MEGA

MOSI------------51
SCK-------------52
MISO-----------50
NSS-------------10
RESET-----------9
DIO0-------------2

torntrousers, 2018-11-28:
@Flavio-da-Brescia do you have a setPins call in your code for the Mega pins? 
```LoRa.setPins(10, 9, 2)```

Flavio-da-Brescia, 2018-11-28:
Hello Torntrousers, yes I did try that too, but with no luck.
With Arduino UNO that piece of code works, I can move the pins 10 and 9 to other pins and it works.
I tried 3 different Arduino Mega and none of them will work...
Thank you for the reply.

Flavio-da-Brescia, 2018-11-28:
`/*
 
The transmitter code arduino for sending data packet lora radio module 433MHz  ra02 lora
 
*/
 
#include <SPI.h> // arduino spi library
 
#include <LoRa.h> // arduino libraries ra02 lora

int counter = 0;
 
void setup() {

 LoRa.setPins(10, 9, 2);//NSS, NRESET, and DIO0 pins can be changed by using LoRa.setPins(ss, reset, dio0).
  Serial.begin(9600);
 
  while (!Serial);
 
  Serial.println("LoRa Sender");
   //LoRa.setPins(10, 9, 2);

 
  if (!LoRa.begin(433E6)) {
 
    Serial.println("Starting LoRa failed!");
 
    while (1);
 
  }
 
 LoRa.setSpreadingFactor(10);
 
 LoRa.setSignalBandwidth(62.5E3);
 
 LoRa.crc();

 
 
}
 
void loop() {
 
  Serial.print("Sending packet: ");
 
  Serial.println(counter);
 
   // send packet
 
  LoRa.beginPacket();
 
  LoRa.print("Halo Like Circuit");
 
  Serial.println(counter);
 
  LoRa.endPacket();
 
  counter++;
 
  delay(2000);
 
}`

This is the code.

Im also using logic level shift, for UNO and for MEGA but no luck with the MEGA, only.

Flavio-da-Brescia, 2018-11-29:
Ok I solved it. I simply forgot to connect the 5V of the logic converter. My bad. Thank you very much for all your support

closed 2020-01-28 by morganrallen