sandeepmistry/arduino-LoRa #193: Starting LoRa failed! issue

shantrekster, 2018-10-03:
Hi ,
I am Shantanu. I only get on my screen:

LoRa Sender
Starting LoRa failed!

I have no idea as to why this happens. Can somebody help me out?
My code is:
#include <SPI.h>
#include <LoRa.h>

int counter = 0;

void setup() {
Serial.begin(9600);
while (!Serial);

Serial.println("LoRa Sender");

if (!LoRa.begin(433E6)) {
Serial.println("Starting LoRa failed!");
while (1);
}
}

void loop() {
Serial.print("Sending packet: ");
Serial.println(counter);

// send packet
LoRa.beginPacket();
LoRa.print("hello ");
LoRa.print(counter);
LoRa.endPacket();

counter++;

delay(5000);
}

wero1414, 2018-10-22:
What hardware setup are you using?

phuongnam0907, 2018-10-24:
> 
> 
> What hardware setup are you using?

I used Arduino Uno R3

morganrallen, 2018-10-25:
@phuongnam0907 You almost certainly need to call `LoRa.setPins`

See https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-pins

wero1414, 2018-10-29:
Yes you need to change the pins of the SS the RST and the INT0  in order to get your module work

nabilballoumi, 2019-08-30:
hello Mr Morganrallen,
help me please   
Can somebody help me out ?

*********************************************************code LORA  Receiver
#include <SPI.h>
#include <LoRa.h>
#include <MKRWAN.h>
LoRaModem modem;
void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Receiver");

    if (!modem.begin(US915)) {

    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    Serial.print("Received packet '");

    // read packet
    while (LoRa.available()) {
      Serial.print(LoRa.read());
    }

    // print RSSI of packet
     Serial.print("' with RSSI ");
    Serial.println(LoRa.packetRssi());
    delay(5000);
  }
 
}
*********************************************************code LORA   Sender
#include <SPI.h>
#include <LoRa.h>
#include <MKRWAN.h>

int counter = 0;

LoRaModem modem;

void setup() {
  Serial.begin(115200);
  while (!Serial);

// No need to call modem.dumb() with arduino-LoRa >= 0.5.0
// modem.dumb();

  Serial.println("LoRa Sender");

  // Configure LoRa module to transmit and receive at 915MHz (915*10^6) 
  // Replace 915E6 with the frequency you need (eg. 433E6 for 433MHz)  US915
  if (!modem.begin(US915)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;

  delay(5000);
}

*********************************************************COM  LORA  Receiver
11:58:23.175 -> LoRa Receiver
11:58:24.023 -> Received packet '0316303125500255255025531.........

*********************************************************COM  LORA  Sender

12:03:50.525 -> LoRa Sender
12:03:51.370 -> Sending packet: 0
12:03:56.371 -> Sending packet: 1
12:04:01.371 -> Sending packet: 2
12:04:06.364 -> Sending packet: 3

2 divice    Arduino MKR wan 1300

morganrallen, 2019-08-30:
You are mixing and matching two different libraries.

```
LoRaModem modem;
```

This is part of MKRWAN (not this library). Then you're calling `LoRa.beginPacket();` which is part of this library but you have not called `LoRa.begin`

pazmat, 2019-09-22:
I am stuck with the same issue of lora starting failed. I have tried quite a few things but failing again and again. My LoRa module doesn't have any pin specified as ss, reset or dio how do i do this. Check the attachment to see the module i am using and help. I had made it work wth softserial though but now after a few days even softwareserial is not working.. I need help urgently. Thanks
[image]

AyeshaYasin94, 2020-04-15:
Hi,
I am new to this .  I am getting error "Starting LoRa Failed" at transmission side.
I am using 2 Aurdino UNO+ 2 433MHZ SX1278 Lora for sending and receiving soil moisture sensor value.   i interface transmitting and receiving  end to aurdino  in this way 
[image]
[image]

I followed the code given in this link
https://how2electronics.com/sending-sensor-data-lora-sx1278-arduino/

Can any one please help me?

IoTThinks, 2020-04-15:
Try this before LoRa.begin() in setup
``
pinMode(LORA_SS, OUTPUT);
digitalWrite(LORA_SS, HIGH);

LoRa.setPins(LORA_SS, LORA_RST, LORA_DIO0);
``

AyeshaYasin94, 2020-04-24:
> Try this before LoRa.begin() in setup
> Replace LoRa_XX with your actual GPIOs.
> 
> ```
> pinMode(LORA_SS, OUTPUT);
> digitalWrite(LORA_SS, HIGH);
> 
> LoRa.setPins(LORA_SS, LORA_RST, LORA_DIO0);
> ```

Hi i used this in Transmitter and Receiver Code as below, But still no success

#define LORA_SS 10
  #define LORA_RST 9
  #define LORA_DIO0 2

void setup() {

      pinMode(LORA_SS, OUTPUT);
digitalWrite(LORA_SS, HIGH);

LoRa.setPins(LORA_SS, LORA_RST, LORA_DIO0);

  Serial.begin(9600);
.....

}

IoTThinks, 2020-04-25:
Your wiring looks fine for me.
You may try to use external 3.3v power for LoRa. 
And/or reduce to SPI Frequency to 4Mhz or 1Mhz 
```
LoRa.setSPIFrequency(1E6);
LoRa.setSPIFrequency(4E6);
```

https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-spi-frequency

AyeshaYasin94, 2020-04-29:
no change :(
How to test that there is no issue in lora chip (i mean your lora chip not
broken).
to test that the issue is not  in the hardware , It's failing because of
code.

Any suggestion?