sandeepmistry/arduino-LoRa #194: Channel selection (or: specific frequency)

MITEL-UNIUD, 2018-10-04:
Hi,
I am trying to make the LoRa Receiver example to function, using as a test sender a node made for TTN LoRaWAN with ABP (I do not care about  content, just testing). Pin setup seems ok. 
From the API description, it seems that I have to select the frequency with `LoRa.setFrequency(frequency)`,  where frequency, from API documentation, is among `433E6, 866E6, 915E6`. I have two questions regarding this: is `866E6` a mistake for `868E6`, and then, if I want to listen on a specific frequency, may I set it directly (e.g., with `868100000`)? 
Then I set spreading factor.
I have set Sync word to 0x34 according to Semtech documentation.  
Default coding rate (4/5) and default  bandwidth (125) are ok.
What else do I need to set? I still do not see received messages. 
Thanks (for the library and for the answer).

MITEL-UNIUD, 2018-10-05:
Self-answers: 
- I discovered that I cannot see received messages because, for some reason, the Serial Monitor on the Arduino IDE stops showing things after printing "LoRa Receiver". Connected to an external Terminal, it shows received messages. 
- yes, I can set the specific frequency, and 866E6 is a mistake (but not crucial, since it is possible to set any frequency).

ricaun, 2018-10-05:
Hello,

This library is not intended to use with LoRaWAN, but is possible to sniff some payloads...
First, you need to check the frequency plan (https://www.thethingsnetwork.org/docs/lorawan/frequency-plans.html).
I guess your plan is EU863-870.
Default we have set Sync word to 0x34, coding rate 4/5;

Let's try channel 1!

**Uplink Channel 1: 868.1 - SF7BW125 to SF12BW125**
Set the frequency to 868100000 and SF7BW125 (Some codes randomize between the SF7BW125 to SF12BW125).

**RX1 Downlink Channel 1: Same as the uplink** 
In this case, is the same frequency, spreading factor, and bandwidth. 
And you need to Invert IQ Signals (Check the enableInvertIQ function). 
Usually, the gateway sends the downlink after 1 second from the uplink when is ABP.

**RX2 Downlink Channel 1: 869.525 - SF9BW125 (RX2 downlink only)**
The Invert IQ Signals still enable, set the frequency to 869.525 and SF9BW125.
The gateway sends the downlink RX2 after 2 seconds from the uplink.

That's it, this is the logic for Uplink and Downlink.
But remember the LoRaWAN payload have a lot of encryption craziness and others tricks...

Stay tuned!

MITEL-UNIUD, 2018-10-05:
Thanks, as you see from the previous message I was finally able to receive messages from a TTN node (actually, I was already receiving but not seeing them in the serial monitor). However, thanks for the InvertIQ explanation, I did not yet consider it.
I know this library is not intended for LoRaWAN, and having already LMIC, in principle there is no strong need for others (althiugh LMIC is not Class C capable...).

closed 2020-11-12 by IoTThinks