sandeepmistry/arduino-LoRa #195: LoRa.h +Teensy3.5 = error

reyupl, 2018-10-09:
I compile the example from the https://github.com/sandeepmistry/arduino-LoRa library witch Teensy LC and works. After changing the plate to Teensy 3.5 I got errors. Can anyone check what is wrong? :
    //codebegin
    /*
    LoRa register dump
    This examples shows how to inspect and output the LoRa radio's
    registers on the Serial interface
    */
    #include <SPI.h> // include libraries
    #include <LoRa.h>

    void setup() {
    Serial.begin(9600); // initialize serial
    while (!Serial);

    Serial.println("LoRa Dump Registers");

    // override the default CS, reset, and IRQ pins (optional)
    // LoRa.setPins(10, 7, 8); // set CS, reset, IRQ pin

    if (!LoRa.begin(915E6)) { // initialize ratio at 915 MHz
    Serial.println("LoRa init failed. Check your connections.");
    while (true); // if failed, do nothing
    }

    LoRa.dumpRegisters(Serial);
    }

    void loop() {
    }
    //end code
    error when compile :

    Arduino:1.8.5 (Windows 10), TD: 1.42, Płytka:"Teensy 3.5, Serial, 120 MHz, Faster, US English"

    C:\Users\rey\Documents\Arduino\libraries\LoRa\src\ LoRa.cpp: In member function 'void LoRaClass:nReceive(void (*)(int))':

    C:\Users\rey\Documents\Arduino\libraries\LoRa\src\ LoRa.cpp:333:55: error: invalid conversion from 'int' to 'IRQ_NUMBER_t' [-fpermissive]

    SPI.notUsingInterrupt(digitalPinToInterrupt(_dio0) );

    ^

    In file included from C:\Users\rey\Documents\Arduino\libraries\LoRa\src/LoRa.h:8:0,

    from C:\Users\rey\Documents\Arduino\libraries\LoRa\src\ LoRa.cpp:4:

    C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SPI/SPI.h:483:7: note: initializing argument 1 of 'void SPIClass::notUsingInterrupt(IRQ_NUMBER_t)'

    void notUsingInterrupt(IRQ_NUMBER_t interruptName);

    ^

    C:\Users\rey\Documents\Arduino\libraries\LoRa\src\ LoRa.cpp: In member function 'long int LoRaClass::getSignalBandwidth()':

    C:\Users\rey\Documents\Arduino\libraries\LoRa\src\ LoRa.cpp:436:1: warning: control reaches end of non-void function [-Wreturn-type]

    }

    ^
    --
    regard Rafal

gitcnd, 2018-11-03:
maybe change line 333 to this:

    SPI.notUsingInterrupt((IRQ_NUMBER_t)digitalPinToInterrupt(_dio0)); // cnd - original

gitcnd, 2018-11-03:
Yep - that seems to work OK.  (my DIO0 connected to teensy pin 9 if that makes a difference)

I also added this at line 435 (after the last case) to get rid of that void warning:
     default : return -1; // cnd - added this

Note! - I see that the files in this github are newer than the files you get by installing via the arduino tools menu... no idea if that's important

gitcnd, 2018-11-03:
I get this on my Teensy 3.2:-
```

LoRa Dump Registers
0x0: 0xEB
0x1: 0x81
0x2: 0x1A
0x3: 0xB
0x4: 0x0
0x5: 0x52
0x6: 0x6C
0x7: 0x40
0x8: 0x0
0x9: 0x8F
0xA: 0x9
0xB: 0x2B
0xC: 0x23
0xD: 0x2
0xE: 0x0
0xF: 0x0
0x10: 0x0
0x11: 0x0
0x12: 0x0
0x13: 0x0
0x14: 0x0
0x15: 0x0
0x16: 0x0
0x17: 0x0
0x18: 0x10
0x19: 0x0
0x1A: 0x0
0x1B: 0x0
0x1C: 0x0
0x1D: 0x72
0x1E: 0x70
0x1F: 0x64
0x20: 0x0
0x21: 0x8
0x22: 0x1
0x23: 0xFF
0x24: 0x0
0x25: 0x0
0x26: 0x4
0x27: 0x0
0x28: 0x0
0x29: 0x0
0x2A: 0x0
0x2B: 0x0
0x2C: 0x0
0x2D: 0x50
0x2E: 0x14
0x2F: 0x45
0x30: 0x55
0x31: 0xC3
0x32: 0x5
0x33: 0x27
0x34: 0x1C
0x35: 0xA
0x36: 0x3
0x37: 0xA
0x38: 0x42
0x39: 0x12
0x3A: 0x52
0x3B: 0x1D
0x3C: 0x0
0x3D: 0xAF
0x3E: 0x0
0x3F: 0x0
0x40: 0x0
0x41: 0x0
0x42: 0x12
0x43: 0x24
0x44: 0x2D
0x45: 0x0
0x46: 0x3
0x47: 0x0
0x48: 0x4
0x49: 0x23
0x4A: 0x0
0x4B: 0x9
0x4C: 0x5
0x4D: 0x84
0x4E: 0x32
0x4F: 0x2B
0x50: 0x14
0x51: 0x0
0x52: 0x0
0x53: 0x10
0x54: 0x0
0x55: 0x0
0x56: 0x0
0x57: 0xF
0x58: 0xE0
0x59: 0x0
0x5A: 0xC
0x5B: 0xF6
0x5C: 0x8
0x5D: 0x0
0x5E: 0x5C
0x5F: 0x78
0x60: 0x0
0x61: 0x1C
0x62: 0xE
0x63: 0x5B
0x64: 0xCC
0x65: 0x0
0x66: 0x1
0x67: 0x50
0x68: 0x0
0x69: 0x0
0x6A: 0x0
0x6B: 0x0
0x6C: 0x0
0x6D: 0x0
0x6E: 0x0
0x6F: 0xB
0x70: 0xD0
0x71: 0x0
0x72: 0x14
0x73: 0x0
0x74: 0x0
0x75: 0x0
0x76: 0x0
0x77: 0x0
0x78: 0x0
0x79: 0x0
0x7A: 0x0
0x7B: 0x0
0x7C: 0x0
0x7D: 0x0
0x7E: 0x0
0x7F: 0x0

```