sandeepmistry/arduino-LoRa #196: CRC - How to check it on the Receiver?

saramonteiro, 2018-10-23:
Hi, 
I am starting to use LoRa.h right now, and I enabling CRC in the transmitter in the setup function. But how should I check it on the Receiver?
thanks in advance!

morganrallen, 2020-01-28:
Closing this as it's known there is no way to be notified when a CRC error occurs. A feature request has been outlined in #326

closed 2020-01-28 by morganrallen

mcgurk, 2020-02-08:
> The library doesn't currently do a lot with CRC check aside checking if the radio set the CRC Error flag. So, if you've called `enableCrc` on **both** the transmitter and the receiver, you should either receive the packet if there is no error, else `parsePacket` will return 0.

`enableCrc` does nothing when used with receiver. Datasheet page 30: "On the receiver side, the bit RxPayloadCrcOn in the register RegModemConfig1 is not used"

morganrallen, 2020-02-08:
This was a great catch @mcgurk, I will be pulling in your PR next week.