sandeepmistry/arduino-LoRa #202: Any way to Continous Receive on MKR1300?

Deltoz, 2018-11-16:
Hello.
Thanks you so much for the work you are doing with your library.
I have been using it for a while to communicate between 2 ARDUINO MKR WAN1300, a transmitter and a receiver.
It works very well, however, as I understand it, the continuous reception function is not compatible with this hardware.
Unfortunately I have the problem that if the transmitter sends a code while the receiver is working on something else ... this code sent is not received by the receiver and I lost is.
is there no way to solve this?
Is it possible that a trivial BT module has its own buffer in reception and this LORA module (more sophisticated) does not have its own reception buffer that can be used?
I hope to find a solution to this problem.
Thanks a lot.

chizran, 2019-05-22:
Sorry to revive an old post, I would like to use 1 MKR WAN1300 as a gateway and for starters 2 MKR WAN1300  as nodes and the issue is lack of Continous Receive on MKR WAN1300.  Looking at the example LoRaSimpleGateway uses receive for rxMode.
 `void LoRa_rxMode(){
  LoRa.disableInvertIQ();               // normal mode
  LoRa.receive();                       // set receive mode
}

void LoRa_txMode(){
  LoRa.idle();                          // set standby mode
  LoRa.enableInvertIQ();                // active invert I and Q signals
}`

So my question is how to either enable Continous Receive for the board or any way to set one board as gateway and the others as nodes?

wero1414, 2019-05-23:
You only can set one as a receiver and check the FIFO or the IRQ registers which is the way that this library works for the MKR1300 so yo only need to set the frequency and set your "Gateway" as RX and every "Node" as Tx