sandeepmistry/arduino-LoRa #204: transmits a single package and stop

alemori, 2018-11-20:
Hello, I'm using an arduino one connected to a sx1278 of 433mhz, when I run the shipping sketch runs the first package and then does not send any more, what can be?

morganrallen, 2018-11-20:
What is "the shipping sketch"? If you mean the sketch the device shipped with, maybe that's all it does? Try working with one of the examples provided.

alemori, 2018-11-21:
I am using your sketch provided by you

wero1414, 2018-11-22:
Please provide your code and your wiring to the SX1278

morganrallen, 2018-11-23:
Two things stand out right away.
1) you're not using the default pins you are **not** calling `setPins`
This has me surprised you are receiving any packets at all

2) though your example cuts off so I cannot be sure, if you don't have a delay in your `loop` you'll be sending the packet over and over again as fast as the processor can handle, this will prevent the radio from actually ever transmitting.

I think it's more likely the first issue, you need to call `setPins`

alemori, 2018-11-23:
Sorry, the scketch has setpins(15,5,4) with the corresponding pins according to the connection you send, there is also the delay. I copied the sketch badly.