sandeepmistry/arduino-LoRa #205: Arduino MEGA getting "starting LORA Failed"

Flavio-da-Brescia, 2018-11-28:
Hello and thank you in advance for the help. I have Arduino Mega and Lora-02 module.

The Module works fine with Arduino UNO, as I tried, but when I wire it to the Arduino Mega it is not able to communicate with the module.

I wired it like this:

LORA------------MEGA

MOSI------------51
SCK-------------52
MISO-----------50
NSS-------------10
RESET-----------9
DIO0-------------2

I am using SPI.h and LoRa.h Library

What can be the reason? Thank you.

morganrallen, 2018-11-28:
Are you calling `setPins`?
https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-pins

Flavio-da-Brescia, 2018-11-29:
Ok I solved it. I simply forgot to connect the 5V of the logic converter. My bad. Thank you very much for all your support

ranaahtishamali, 2019-11-04:
> 
> 
> Ok I solved it. I simply forgot to connect the 5V of the logic converter. My bad. Thank you very much for all your support

Please can you upload the circuit diagram of it. its not working!

feras33, 2020-03-19:
hello everyone 
i am using lora sx1278 ra-02 , and i have tried this method with connecting exactly the same as mention above .Unfortunately it still not working i dont know why can any one help ?.

IoTThinks, 2020-03-20:
@feras33 Why dont you create a new thread?

ranaahtishamali, 2020-03-24:
> 
> 
> hello everyone
> i am using lora sx1278 ra-02 , and i have tried this method with connecting exactly the same as mention above . Unfortunately it still not working i dont know why can any one help ?.

Send your circuit diagram.

feras33, 2020-03-24:
I wired it like this:

LORA------------MEGA

MOSI------------51
SCK-------------52
MISO-----------50
NSS-------------10
RESET-----------9
DIO0-------------2

I am using SPI.h and LoRa.h Library
Same as it is mentioned in this thread

ranaahtishamali, 2020-03-25:
Attach DIO1 to 3.
Is your lora module of 433mhz?

feras33, 2020-03-25:
Yes it is 433MHZ

feras33, 2020-03-25:
I have changed the frequency in the code from the library

agisyahrain, 2020-04-15:
do you have source code for the receiver? and using arduino mega too?

IoTThinks, 2020-04-15:
Your mapping to SS to 10 may be wrong.
And add the two magic lines:
```
// SS, RST, DIO0
  pinMode(LORA_SS, OUTPUT);
  digitalWrite(LORA_SS, HIGH);  
```

[image]

Full of my code

> #include <SPI.h> // arduino spi library
> #include <LoRa.h> // arduino libraries ra02 lora
> 
> #define LORA_SS   53
> #define LORA_RST  9
> #define LORA_DIO0 9
> 
> int counter = 0;
> void setup() {
>   // SS, RST, DIO0
>   pinMode(LORA_SS, OUTPUT);
>   digitalWrite(LORA_SS, HIGH);  
>   
>   LoRa.setPins(LORA_SS, LORA_RST, LORA_DIO0);
>   Serial.begin(115200);
>  
>   while (!Serial);
>   Serial.println("LoRa Sender");
> 
>   if (!LoRa.begin(433E6)) {
>     Serial.println("Starting LoRa failed!");
>     while (1);
>    }
>  
>   // LoRa.setSpreadingFactor(10);
>   // LoRa.setSignalBandwidth(62.5E3);
> }
>  
> void loop() {
>   Serial.print("Sending packet: ");
>   Serial.println(counter);
>  
>   // send packet
>   LoRa.beginPacket();
>   LoRa.print("Halo Like Circuit");
>   Serial.println(counter);
>   LoRa.endPacket();
>   counter++;
>   delay(2000);
>  
> }

IoTThinks, 2020-04-15:
I'm too lazy to write a receiver so I turn on my SDR.
Something is sending at 433Mhz. in the middle.
Hope it works for you guys.

[image]

IoTThinks, 2020-04-15:
Overview of my physical wiring.
[image]

alfantamamy, 2020-04-16:
> Overv- iew of my physical wiring.
> [image]
> 

I tried too, but still not working. I used arduino mega as transmitter, and uno as receiver. Both are using lora.

alfantamamy, 2020-04-16:
> Overview of my physical wiring.
> [image]

[image]
my physical wiring

IoTThinks, 2020-04-17:
As an IT, define "not working".

alfantamamy, 2020-04-17:
> As an IT, define "not working".

Starting Lora Failed

IoTThinks, 2020-04-17:
Did you put this in front of LoRa.begin()
```
 pinMode(LORA_SS, OUTPUT);
 digitalWrite(LORA_SS, HIGH); 
```

If still failed, your LoRa is likely faulty.
Beware that LoRa is 3.3v.

alfantamamy, 2020-04-17:
> Did you put this in front of LoRa.begin()
> ```
>  pinMode(LORA_SS, OUTPUT);
>  digitalWrite(LORA_SS, HIGH); 
> ```
> 
> If still failed, your LoRa is likely faulty.
> Beware that LoRa is 3.3v.

Already, previously I tried to connect LoRa with arduino uno, it works to sending data, which is my LoRa is good. But I tried to connect arduino mega, failed.

*I copied your code, and your set pins to my project

IoTThinks, 2020-04-17:
What GPIO you use to connect to NSS/SS of LoRA?

alfantamamy, 2020-04-17:
Digital pin 53

IoTThinks, 2020-04-17:
Try to use external 3.3v for LoRa
If still failed then you have to rely on your own already. 
Mostly like some error on your side or your board.

JanRealAcc, 2023-03-24:
Hehe just found out the solution to this problem. It was LoRa.h that make this error. Just change the SS 10 to 53 if you wish to use arduino mega. 

[image]

There's another method, but I prefer the 1st method.

[image]

Hope it helps :)