sandeepmistry/arduino-LoRa #207: Modulation Mode Selection

jctang46, 2018-11-30:
Hi Sandeep,
For using SX1276, if the default modulation mode is LoRa (CSS)?It supports FSK, GFSK, MSK, GMSK, LoRa and OOK. If not how to select (config) modulation mode? Before I use this library, first I need make sure I am in LoRa mode, if I only want to use a unique one among them. If you have any function in this library can allow me to set modulation mode? I know I can use dumpRegisters or readRegister to check the current modulation mode, if you can provide functions  to check and set/reset modulation mode it would be better.

manvendra88, 2019-05-28:
Hi jctang46,
Did you make up your own function for this? As I couldn't find one in the library.

torntrousers, 2019-05-28:
This library only supports LoRa mode and internally it sets this mode to always be on, eg https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L385

manvendra88, 2019-05-28:
Thanks a lot, torntrousers for pointing it out.
https://github.com/Lora-net/LoRaMac-node/blob/develop/src/radio/sx1276/sx1276.c#L407
I think this one provides the selection of the mode.

jctang46, 2019-05-28:
Hi Manvendra,
Not yet. What I have done is a GPS-LoRa tracking system. I used a Neo-6M GPS to get the location, display it on an OLED , at the same time transfer it to the receiving side by LoRa (SX 1276). The receiving side LoRa decodes the info , also displays on an OLED , at the same time uses WiFi (the microprocessor ESP32 has its own) to “publish” it to PubNub (my channel) , by using EON map , anyone in the world who knows my PubNub “subscribe”  key and channel can find my real time location on EON map. The test was done . The SX1276 with 100 mw power can cover an area with one mile of radian in a residential area. (antenna 2 dbi for both sides) If put the transmission side on an UAV , it could cover an area for tens of miles. What I want is a LoRa lib can provide functions to switch SX1276 among Semtech LoRa mode and traditional FSK, GFSK etc.

manvendra88, 2019-05-28:
> Hi Manvendra, Not yet. What I have done is a GPS-LoRa tracking system. I used a Neo-6M GPS to get the location, display it on an OLED , at the same time transfer it to the receiving side by LoRa (SX [...]

Hi jctang,
Great job. :)) 
Curious to know about your project in greater detail (i will dm you for that). :))
For the library part, as @torntrousers has mentioned that this library uses LoRa mode by default, i happened to search a few libraries that support modification of different modes using https://www.youtube.com/watch?v=0dnnqToK28A&t=68s
and found this one suitable for the job https://github.com/Lora-net/LoRaMac-node/blob/develop/src/radio/sx1276/sx1276.c#L407.