sandeepmistry/arduino-LoRa #208: Set frequencies between 865 - 867 MHz

hussainb, 2018-11-30:
Hi,
I am exploring LoRa and wish to get few SX1278 modules.

I would like to run it in frequency range 865 - 867 MHz.
Looking at your documentation the freq `433E6`, `866E6`, `915E6 ` can only be configured.

I was also looking around the web and saw this post [HERE](https://www.libelium.com/forum/viewtopic.php?t=24217)

The loraWAN library can turn off channels using `LoRaWAN.setChannelStatus` and can also set frequency for a channel by `LoRaWAN.setChannelFreq`.

I do not wish to get into LoRaWAN and would like to use this library for individual purposes, but I would like to configure it between 865 - 867 MHz.

I wanted to know if something similar is possible here. If not, is it because of any hardware limitations or its just that its not yet implemented in code.

Thanks!

torntrousers, 2018-12-01:
HI @hussainb , where did you read that only freq 433E6, 866E6, 915E6 can only be configured? You can pass in anything you like on the ```begin``` and ```setFrequency``` call's, eg: ```LoRa.begin(866123456)```. If your Lora module supports the frequency then it should work.

hussainb, 2018-12-01:
Hi @torntrousers,

Thanks for the reply.

I read it here [API.md](https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md)
Being a beginner to this, I just assumed that it supports only those frequencies since they are explicitly mentioned, custom frequencies isn't mentioned. Thanks for clearing that it accepts custom values.

your reply clears the question for setting custom frequency, is it possible to turn off channels through this library the way I mentioned above?

Marcoconti1960, 2019-02-19:
> HI @hussainb , where did you read that only freq 433E6, 866E6, 915E6 can be configured? You can pass in anything you like on the `begin` and `setFrequency` call's, eg: `LoRa.begin(865123456)`. If your Lora module supports the frequency then it should work.

I understand torntrousers' question as I initially had a similar problem with the au_915 frequencies. For some reason, I could initially compile only using 915E6,. Then, I added and removed libraries and now I could compile using  LoRa.begin(915000000). Now I am in the stage to test if the Dragino Lorashield is effectively transmitting at 915Mhz. A clarification about the acceptable parameters in LoRa.begin() would be welcome.

Rita-GE, 2020-08-17:
I have a similar question.
Is SX1278  possible to run frequency hopping function through this lib?
Like @hussainb, I need it can run in the frequency range 865 - 867 MHz.

IoTThinks, 2020-08-17:
If your modules support high frequencies, you can run at any high frequencies, 868, 869, 899....915, 916 ...at no problem.

No need to be fixed at any well-known frequencies.

closed 2021-07-04 by IoTThinks

8G6, 2022-08-07:
> 
SX1278 only supports 410Mhz to 525Mhz