sandeepmistry/arduino-LoRa #212: Not an issue - guidance on Idle mode

3Dtj, 2018-12-18:
Hi guys,

I want to use an Arduino Uno to have my SX1276 LoRa chip go into idle mode and then activate when it receives a preamble.  I see that command for LoRa.idle() to initiate the idle mode but unclear how to re-activate.  Semtech's data sheet discusses the various registers associated with preamble length thus do I need to write a command for the various SX1276 shift registers?

Thank you!

3Dtj, 2018-12-19:
Also, does LoRa.h library support the beacon function?

Happy holidays!

morganrallen, 2018-12-19:
Last part first, beacons are not part of LoRa, as far as I can tell that only applies to running in FSK/OOK mode.

For wake-up you could look into `CAD Detection`. I don't have any hands on experience with this yet but as I understand, if you have `DIO0` **or** `DIO3` and `DIO1` **or** `DIO4` available you should be able to receive interrupts for CAD Detect (when LoRa is sensed) and CAD Done (successful signal demod). 

Unfortunately I don't have any hardware with the other DIO pins exposed so I have be unable to support this feature.

3Dtj, 2018-12-20:
Thanks morganrallen.  Greatly appreciate you always responding so quickly on these questions. 

Just to confirm on the beacon topic, did you deduce that beacon mode is only functional because it is tied to the 0x31 register address associated with available a FSK/OOK?  I'm still learning about the registers but does this mean the LoRa.h library only activates the LoRa functionality for the SX1276 registers?   Guess I'm confused on why there is even FSK/OOK functionality on the SX1276 in the first place.

In regards to CAD, if I want to implement that function do I need to specifically call out the register address (e.g. RegOpMode 0x01) in my Arduino Uno sketch to get it to work?  Is that the case for any other LoRa register functionality that I want to use?

morganrallen, 2018-12-21:
> Just to confirm on the beacon topic, did you deduce that beacon mode is only functional because it is tied to the 0x31 register address associated with available a FSK/OOK? 

That's right. While the datasheet doesn't explicitly state BeaconMode only works in FSK/OOK mode, it's config register (0x31.3) is only found in `Table 42: FSK/OOK Mode Register Map`

> but does this mean the LoRa.h library only activates the LoRa functionality for the SX1276 registers?

Right again, this driver only supports the `LoRa` interface of the `SX127*`. I believe the underlying modulation for LoRa uses some combo of FSK/OOK, so I guess it just made sense to expose that functionality, maybe if you wanted to do your own modulation scheme on top of it? That's mostly speculation, the datasheet may have the answer but I've mostly kept to the LoRa portions of the doc.

> do I need to specifically call out the register address (e.g. RegOpMode 0x01)

By default all of the interrupts are enabled and I do not believe we change that (we haven't even defined RegIrqFlagsMask....) so there should be a call to `handleDio0Rise` for 3 events, `RxDone`, `TxDone` and `CadDone` which only `RxDone` is handled. This PR kind of fell apart, but you should be able to get some guidance from it. https://github.com/sandeepmistry/arduino-LoRa/pull/50

I would love to see another CAD pull request come it, it's something I want but haven't had to the time look into.

3Dtj, 2018-12-24:
Thanks again.  I'll do some more evaluating and see what I can come up with.

closed 2018-12-24 by 3Dtj