sandeepmistry/arduino-LoRa #215: Starting LoRa failed!

zzxjl1, 2019-01-05:
I‘m using an arduino nano 

D12-LEVEL SHIFTER-MISO
D11-LEVER SHIFTER-MOSI
D6-LEVER SHIFTER-NSS
D5-LEVER SHIFTER-RST
D4-LEVER SHIFTER-DIO0
D13-LEVER SHIFTER-SCK

///////////////////////////////////////////////////////
#include <SPI.h>
#include <LoRa.h>

void setup() {
  Serial.begin(115200);
  while (!Serial);

  Serial.println("LoRa Receiver");
LoRa.setSPIFrequency(1E6);
  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    Serial.print("Received packet '");

    // read packet
    while (LoRa.available()) {
      Serial.print((char)LoRa.read());
    }

    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(LoRa.packetRssi());
  }
}

iw2lsi, 2019-01-17:
Hi... 

  take a look at the default port settings:

#define LORA_DEFAULT_SPI           SPI
#define LORA_DEFAULT_SPI_FREQUENCY 8E6 
#define LORA_DEFAULT_SS_PIN        10
#define LORA_DEFAULT_RESET_PIN     9
#define LORA_DEFAULT_DIO0_PIN      2

am I wrong or you are using differents GPIO ?

see also:

void setPins(int ss = LORA_DEFAULT_SS_PIN, int reset = LORA_DEFAULT_RESET_PIN, int dio0 = LORA_DEFAULT_DIO0_PIN);

   Best Regards,

               Giampaolo

Abdulsametileri, 2019-03-13:
Try 
`LoRa.setSPIFrequency(1E6);`
to change 
`LoRa.setSPIFrequency(4E6);`

edgarbjorntvedt, 2020-09-12:
I keep getting "Starting LoRa failed!" I've tried a lot of different pin configs. Nothing seems to help. I've used the code in /examples/LoRaSender/LoRaSender.ino as a base but changed frequenzy to 433E6 in order to match the frequeny on my module.

I'm using ESP8266/NodeMcu with AI-Thinker Ra-01 Lora module. Has anyone get this setup working?