sandeepmistry/arduino-LoRa #216: ESP-IDF versus Arduino : LoRa.start() fails on ESP32

brettbeeson, 2019-01-12:
Hello! I'm using a ESP32 LoRa board[ TTGO V2.1](https://www.aliexpress.com/item/TTGO-LoRa32-V2-1-1-6-version-868-915Mhz-ESP32-LoRa-OLED-0-96-Inch-SD/32872537745.html), compiling with the latest Arduino and ESP-IDF.

On Arduino, I've got the [LoRaReceiver](https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaReceiver/LoRaReceiver.ino) and Transmitter to work by adding the following to set the pins:

```
LoRa.setPins(18, 23, 26);
    if (!LoRa.begin(915E6)) {
```  
This works fine in Arduino IDE.

On ESP-IDF, using "git clone" to install the library, I can build and upload it. However, with _exactly the same example_ I receive a "Initialization Failed." Checking the code, this occurs because the "readRegister(REG_VERSION);" in LoRA::begin(...) returns 0.

How can I correctly set the library to use ESP-IDF, or how could I debug further?

Thanks, Brett

morganrallen, 2019-01-13:
I'm a little confused about the setup you're describing. This library lacks the necessary config files to integrate into `ESP-IDF`, though you still can use external libraries. If you can describe your build better maybe we can figure this out. 

Alternatively, I also wrote an `ESP-IDF` specific component for LoRa. https://gitlab.com/morganrallen/esp32-lora/

brettbeeson, 2019-01-13:
Thanks for your reply, and assistance.

I've moved to the esp32-lora library as a component in esp32-lora. I installed (as per "git submodule, etc" instructions on your link) fine. Running a simple test program, however, fails with an error (copied below).

Note the board is a [TTGO Lora V2.1_1.6](https://www.aliexpress.com/item/TTGO-LoRa32-V2-1-1-6-version-868-915Mhz-ESP32-LoRa-OLED-0-96-Inch-SD/32872537745.html). I changed the "GPIO to handle RESET" to 23, as per the spec sheet. The other pins seem ok.

```
I (185) scan: Hi
I (185) gpio: GPIO[18]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (195) gpio: GPIO[23]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (205) LoRa32: Toggling reset pin 23
I (405) LoRa32: Bringing NSS high: 18
I (415) LoRa32: Initializing SPI bus
I (415) LoRa32: 
                     MISO: 19
MOSI: 27
CLK: 5
NSS: 18
I (415) LoRa32: lora32_sleep
I (415) LoRa32: REG_FRF_MSB: 0xD8
I (415) LoRa32: REG_FRF_MID: 0x80
I (425) LoRa32: REG_FRF_LSB: 0x 0
I (425) LoRa32: lora32_set_frequency: 866000000
I (435) LoRa32: clear rx/tx fifos
I (435) LoRa32: set lna: 0x23
I (435) LoRa32: set_tx_power(17, 1)
I (445) LoRa32: lora32_set_tx_power
I (445) LoRa32: lora32_idle
I (455) LoRa32: Setting callback handler
I (455) gpio: GPIO[26]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:1 
I (1055) LoRa32: handling DIO0
E (1055) spi_master: check_trans_valid(1093): invalid dev handle
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x40084e5c
0x40084e5c: _esp_error_check_failed at /home/bbeeson/esp/esp-idf/components/esp32/panic.c:712
file: "/home/bbeeson/esp/scan2/components/esp32-lora/main/esp32-lora.c" line 79
func: lora32_read_reg
expression: spi_device_transmit(lora->spi, &t)
Backtrace: 0x40084a18:0x3ffb71d0 0x40084e5f:0x3ffb71f0 0x400e436d:0x3ffb7210 0x400e43fa:0x3ffb7260 0x40086871:0x3ffb7280
0x40084a18: invoke_abort at /home/bbeeson/esp/esp-idf/components/esp32/panic.c:707
0x40084e5f: _esp_error_check_failed at /home/bbeeson/esp/esp-idf/components/esp32/panic.c:713
0x400e436d: lora32_read_reg at /home/bbeeson/esp/scan2/components/esp32-lora/main/esp32-lora.c:309
0x400e43fa: lora32_handle_dio0 at /home/bbeeson/esp/scan2/components/esp32-lora/main/esp32-lora.c:309
0x40086871: vPortTaskWrapper at /home/bbeeson/esp/esp-idf/components/freertos/port.c:403
```
The complete executable, based on your minimal example code is:
```
#include "esp_log.h"
#include "esp_event_loop.h"
#include "esp32-lora.h"

static const char *TAG = "scan";

uint8_t data_available = 0; static lora32_cfg_t lora; static QueueHandle_t uart0_queue;
static void handle_lora_receive(uint8_t *data, uint8_t size) {   printf("msg: %s\n", data); }

void app_main()
{
            ESP_LOGI(TAG, "Hi");
            lora32_cfg_t lora;
  lora = lora32_create();
  lora.receive = &handle_lora_receive;

  lora32_init(&lora);

  lora32_send(&lora, TAG, 5);
}

```

jigsurfer, 2021-08-08:
Hi! How about the encryption for the message send by Lora ! Do you have any idea how to emcrypt the message. I found lot of Lora & LMIC version online, but they are too complicated for my level. Thank you !

IoTThinks, 2021-08-08:
@jigsurfer To create a new topic instead.

The topic was closed and not relevant to your question.