sandeepmistry/arduino-LoRa #219: Compatibility with RAK831 SX1257

janbbeck, 2019-01-23:
First of all, thanks for this library. I used it to get a number of boards talking to one another including the Adafruit 32u4, HELTEC esp32 based boards and even an STM32LRWAN1. They all work great.

However, I cannot get them to talk to a RAK831 concentrator based on the SX1257.
The util_tx_test claims to be sending out the packets sucessfully, but the Arduino boards do not see them
The other way, the util_pkt_logger does not see the packets when sent by the Arduino.

I don't plan to do LoraWAN, I want to grab the packets from the concentrator and then do custom processing on the computer.

Are the two platforms compatible?

Thanks for taking the time to read this.

sabas1080, 2019-01-23:
util_pkt_logger or lora gateway tools only work for loraWAN according to memory, for you experiment should try to directly handle the register of the RAK831 and use only lora

janbbeck, 2019-01-24:
> util_pkt_logger or lora gateway tools only work for loraWAN according to memory, for you experiment should try to directly handle the register of the RAK831 and use only lora

Thank you for taking the time to reply. Using it, I was able to figure it out, though FYI you were incorrect.

I did use a LoraWAN implementation for arduino, and immediately the RAK831 saw the packets. But then I noticed that it was using 904.5 MHz instead of the 915 we put into the sandeep lora code. (Makes a lot of sense in retrospect) Changing using 904.5 Mhz with the sandeep library causes the RAK831 util_pkt_logger to see the packets. 
Thanks again.