sandeepmistry/arduino-LoRa #221: About CAD implementation

sobehard, 2019-01-23:
Hi sandeepmistry, are there any chances that you will implement CAD function for the LoRa receiving side so that it can support multiple spreading factors on a single channel?

3Dtj, 2019-01-23:
I had the same question (issue #212) and actually reached out to the Semtech community forum.  I was told that implementing CAD is easy but unclear if the following solution can be done with Sandeep's Lora.h library.  

Not sure if this makes sense but this is the Semtech representative suggested:
So it's quite easy to perform a CAD on a SX127X, 
You have to configure your radio as usual as is done in this example : https://github.com/Lora-net/LoRaMac-node/blob/develop/src/apps/ping-pong/NucleoL152/main.c line 203/208 

An then you have to declare the On CadDone callback as following : 

// Radio initialization
RadioEvents.TxDone = OnTxDone;
RadioEvents.RxDone = OnRxDone;
RadioEvents.TxTimeout = OnTxTimeout;
RadioEvents.RxTimeout = OnRxTimeout;
RadioEvents.RxError = OnRxError;
RadioEvents.CadDone = OnCadDone; // Add this line

/* Implement you callback  */
void OnCadDone( bool channelActivityDetected)
{
Radio.Sleep( );
/* User app */
    if(channelActivityDetected)
    {
        CadDetected = true;
    }
    else
    {
        CadDetected = false;
    }
State = CAD_DONE;
}

and after that you just have to call the SX1276StartCad(); function (sx1276.c line 1098) when you want perform a CAD.

sobehard, 2019-01-24:
Hi @3Dtj, thanks for the fast response. Actually, I'm kinda new for the LoRa technology, so I have more question about writing the register for the cad mode than writing the cad function. And please correct me if I'm wrong. 
As I know, in order to use the cad function, we have to declare the dio_mapping (1 or 2) and the mode for each dio pins of the LoRa chip example like cad_done, cad_detect and so on according to the datasheet of the LoRa module (I was referring to https://cdn-shop.adafruit.com/product-files/3179/sx1276_77_78_79.pdf - page 46). So if I want to use the cad function, does it means that I will need to use others dio pin instead of just dio 0 for that purpose? If so, how should I handle those dio pins?

3Dtj, 2019-01-24:
Great question and I'm struggling with the same thing.  I've gotten Sandeep's library to work for my SX1279 chips to create a duplex communication but haven't figured out how to implement the built-in CAD functionality.  It seems odd that we have to call out each register and I'm hoping Sandeep or others will chime in on this topic.

morganrallen, 2019-01-25:
Hi just wanted to chime in so you know someone is paying attention.

We've had at least one attempt at implementing CAD detection but the PR fell apart due to having too many other things included.

CAD detection is top of my current priority list as I very much need it in my projects, I just don't have time to dig into it again right now (end of next month, likely)

I would happily work with anyone who submits a PR to get it implemented, thought presently I only have hardware to verify the `CadDone` functionality.

3Dtj, 2019-01-25:
I'm still a novice at the coding but certainly want to assist in the CAD functionality.  I'm trying to round up a few other folks that can assist on it before submitting a PR.  I'll contact my coworkers again and see about making this happen.

3Dtj, 2019-01-25:
Nice find sobehard.  I'll program the sketch this weekend and test it out.

morganrallen, 2019-01-26:
You might find some good info in #50 I'll dig into that myself this weekend if I get a bit of free time.

lazyweirdo, 2020-01-07:
@sandeepmistry any thoughts here ? ... CAD is a very important feature

morganrallen, 2020-01-07:
I would happily merge a pull request. I've started looking into this myself and it looks fairly straight forward, I just don't have time to thoroughly test.

A complete PR should include.

* minimal code to enable & disable CAD detection
* CAD Detect callback similar to `onReceive`
* API documentation
* Examples

PR #50 has most of this but was too complex, though it should provide all the needed detail for anyone who wants to take this PR on.

IoTThinks, 2020-11-12:
Thinking to do a new PR with cad() and onCadDetected()
@morganrallen Have you done the CAD implementation?

unbracedcat500, 2020-12-27:
Hey @3Dtj I am struggling with CAD implementation on my Arduino pro mini. Radiohead library only supports it in Reliable mode & not unreliable mode( the library has 2 modes). So I thought of building the cad function using driver given from semtech github page. However, I see in the ping pong example "radio.h" is invoked which contains pointers to different functions but I couldn't find where these functions are written. Can you help me out with that?

IoTThinks, 2020-12-28:
@unbracedcat500 What is your use case for CAD?
I used to very interested in CAD implementation and about to do a PR myself.

Then I realized that CAD still takes around 1.8mA. So I can not use CAD to wake up MCU during deep sleep. So I feel CAD is quite pointless.

To listen before talk, I use RSSI() to measure noises from other nodes.
Not quite accurate as CAD but still usable.
https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#rssi