sandeepmistry/arduino-LoRa #225: Next release(s)

morganrallen, 2019-02-04:
Hi All! A lot has happened with this library in the last year since I've been working to maintain with @sandeepmistry so I wanted to kick off a conversation about future releases.

Going over https://github.com/sandeepmistry/arduino-LoRa/compare/0.5.0...master

it appears 3 new bit of functionality have been added.

* #62 non blocking functions added
* #179 Add function to invert IQ signal register
* #153 sx127x 20dBm support

I believe people are using the non-blocking functionality and that alone probably warrants a release. I'm proposing tagging a new release (0.6.0) with these changes and starting a checklist for the next.

Things I would like to see in the post-0.6.0 release.

* CAD Detect (#50 and/or #221)
* Address REG_FIFO_ADDR_PTR concerns (#218 #222)
* Update docs if/where needed (#210)

Things I'd like to see in 1.0.0

* Test suite
* ?

Feel free to chime in with any thoughts you have about moving this library forward. As this is a volunteer, community driven effort your feedback will steer future releases.

sandeepmistry, 2019-02-05:
@morganrallen thanks for kicking this off, the plan for a 0.6.0 release sounds good to me!

3Dtj, 2019-02-07:
> Going over 0.5.0...master

I fully support the CAD development but then again I'm biased.

3Dtj

IoTThinks, 2019-11-18:
Hi all,
When will CAD be supported in the roadmap?
I'm quite interested with this feature. :)

BTW, thanks a lot for this awesome library.

morganrallen, 2019-11-18:
Once someone submits a full pull request. Until recently I have not had
hardware that supported it so I haven't done it myself. After the new year
I should have time to do it.

IoTThinks, 2019-11-19:
Thanks a lot Morganrallen.
I could help to test with ESP32 + AI-Thinker Ra02. :)

Ok, take your time and wait for your good news.
I appreciate your time and effort.